bblanchon/ArduinoJson #629: Port for other microcontrollers outside Arduino

passoswell, 2017-12-01:
Hello,

I would like to know if it's possible to use your library on other compilers like Atollic for STM32, Keil uvision for Atmel ARM and 8051, and MPLAB for PICs. I've been looking for a library like ArduinoJson, but none are designed to encode and decode on embedded systems.

Thanks in advance.

bblanchon, 2017-12-02:
Hi @passoswell,

ArduinoJson is very portable and works on virtually any C++ compiler.
The only exception so far is [Embarcadero C++ Builder](https://blog.benoitblanchon.fr/embarcardero-cxx-builder-is-a-joke/), but can we really call this a C++ compiler...

I tested with MPLAB for PIC and the code compiler without any error or warning.
I just needed to define `ARDUINOJSON_EMBEDDED_MODE` to disable usage of the STL:

```c++
#define ARDUINOJSON_EMBEDDED_MODE 1
#include "ArduinoJson-v5.11.2.h"
```

A long-term solution would be to improve the detection in [`ArduinoJson/Configuration.hpp`](https://github.com/bblanchon/ArduinoJson/blob/d3a1203782856bf8907961e6b0212db60d40364d/src/ArduinoJson/Configuration.hpp#L7-L14). I invite you to make a pull request if you're interested.

I didn't have the time to test other compilers you mentioned; please give it a try and tell me how it goes.

Regards,
Benoit

bblanchon, 2017-12-02:
I've just tested with Atollic for ARM and got the same result: zero error or warning.

passoswell, 2017-12-02:
Thanks for your reply!

you said you included the "ArduinoJson-v5.11.2.h" file, but I couldn't find it. The closest thing is ["ArduinoJson.h"](https://github.com/bblanchon/ArduinoJson/blob/master/ArduinoJson.h), and I get 156 compilation errors. Some of then say "unkown type name 'namespace'" pointing to multiple files where namespace appears.

I'm using CubeMX to setup the configuration scheme for my STM32 microcontroller and using the Atollic to code and compile. 

Are you using a different ArduinoJson library, or a different tool setup?

bblanchon, 2017-12-03:
Hi @passoswell,

`ArduinoJson-v5.11.2.h` is the single file distribution of ArduinoJson.
You can download it here: https://github.com/bblanchon/ArduinoJson/releases/latest
You just need to copy that file in your project folder, alongside with your `.cpp` files.

`unknown type name 'namespace'` means that you're compiling in C, whereas C++ is required.
Indeed ArduinoJson is a C++ library so it needs a C++ compiler.
To make it work with Atollic, the file that includes ArduinoJson must have the extension `.cpp`.

Regards,
Benoit

bblanchon, 2017-12-05:
I've just tested with Keil µVision for ARM and got the same result: zero error or warning.

passoswell, 2017-12-05:
Hello, @bblanchon 

I followed your steps and got 0 errors, as you said it would be.

I haven't tested the library, though, but there should be no problems from now on.

Thanks for your replies, your tips worked pretty well.

bblanchon, 2017-12-11:
You're welcome, thanks for your interest in ArduinoJson.
Please come back, and tell us how your project went.

closed 2017-12-11 by bblanchon