bblanchon/ArduinoJson #630: parseObject() failed

sgofferj, 2017-12-03:
Trying to parse this URL on an ESP8266 but parseObject() fails:
http://services.swpc.noaa.gov/products/noaa-estimated-planetary-k-index-1-minute.json

beicnet, 2017-12-03:
You are trying to Parse too big json file, that large ammount of data can not be parsed, I think!

bblanchon, 2017-12-04:
Hi @sgofferj,

You need to call [`parseArray()`](http://arduinojson.org/api/jsonbuffer/parsearray/) as the JSON document contains an array.

Also, the [ArduinoJson Assistant](https://arduinojson.org/assistant/) says you need 30KB to parse this document.
If you don't have enough free memory, consider parsing the input in chunks by calling `parseArray()` repeatedly.

See also:

* [Can I parse a JSON input that is too big to fit in memory?](http://arduinojson.org/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/)

Regards,
Benoit

sgofferj, 2017-12-06:
The ESP8266 should have enough RAM to handle that but I found another way. I first send a HEAD now to get the Content-Length header and then request Ranges. That way, I can get only the last record (that's what I'm interested in) and then I parse it by hand. Not sure, but I think, there might also be a memory hole in the library because the ESP8266 sometimes crashes with a stack dump. That could be the main problem here.

bblanchon, 2017-12-11:
Hi sgofferj,

I'm pretty confident that there is no memory leak in the library.
However, you may get this impression if, against all advice, you reuse the same `JsonBuffer` several times.

See:

* [I found a memory leak in the library!](https://arduinojson.org/faq/i-found-a-memory-leak-in-the-library/)
* [Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/faq/why-shouldnt-i-use-a-global-jsonbuffer/)

Regards,
Benoit

closed 2017-12-11 by bblanchon