bblanchon/ArduinoJson #632: Change the default value not working?

marshallbrown123, 2017-12-10:
Hi there - thanks for the awesome library, I purchased your book to help the cause!

Page 53 or the book has the useful option to change the default value if a field is missing in the JSON Object

example
short tcpPort = config["port"] | 80;

my usage
hour = json_cal_entry["hour"] | WILDCARD;  //if the cal entry is missing from the object then load the wild card.

The bottom of the page states. 

"This syntax is new in ArduinoJson 5.12, and it’s only available when you use the subscript syntax
([])."

Question is:
Where is ArduinoJson 5.12 - I downloaded the latest version at 5.11.2 but the function is not there? 

Thanks

marshallbrown123, 2017-12-10:
I read the change log and see it hasn't been released yet. 
I'll get the head version from GitHub and give it a go!

closed 2017-12-10 by marshallbrown123

marshallbrown123, 2017-12-10:
Yes it works great! thanks

bblanchon, 2017-12-11:
Hi @marshallbrown123,

Sorry about that, all my energy went into writing [the book](https://leanpub.com/arduinojson).
Thank you very much for buying it 👍 

[ArduinoJson 5.12](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.12.0) is out now.
Enjoy!