bblanchon/ArduinoJson #633: parse a primitive

vortex314, 2017-12-11:
I didn't find in the documentation a way to parse a primitive like string,boolean,number which could return a JsonVariant. I have a use case where I can receive any of the JSON primitive values.
The only way around this would be to encapsulate it first as an array and extract element 0.

Any other recommendation ? 

BTW : Great library ! Works excellent on ESP8266 and ESP32.

bblanchon, 2017-12-12:
Hi @vortex314,

You are looking for [`JsonBuffer::parse()`](http://arduinojson.org/api/jsonbuffer/parse/).

Thank you very much for the kind words ;-)

Regards,
Benoit

vortex314, 2017-12-12:
Benoit, thanks for the swift response ! Apparently I succeeded to miss this in the documentation. :-(
Is there also an equivalent to do the opposite like JsonVariant::printTo() ?
Lieven

bblanchon, 2017-12-13:
Hi Lieven,

There is [`JsonVariant::printTo()`](https://arduinojson.org/api/jsonvariant/printto/).
It was missing in the documentation, I just added it.

Regards,
Benoit

vortex314, 2017-12-14:
Hi Benoit,
great ! Thanks for the swift help you provided.
Lieven

closed 2017-12-14 by vortex314

bblanchon, 2017-12-14:
You're welcome, thanks for using ArduinoJson.