bblanchon/ArduinoJson #634: error with dynamic keynames on runtime

jliegner, 2017-12-11:
``````
#include <iostream>
#include "ArduinoJson.h"

int main()
{
  DynamicJsonBuffer aJb;
 
  JsonObject& root = aJb.createObject();
  for (int i=0; i<100;  i++)
    {
    char ca[16];
    sprintf(ca, "%i", i);
    root.set(ca, i);
    }

  size_t len=root.measureLength();
  char *pc=new char[len+1];
  root.printTo(pc, len+1);
  printf("%s", pc);
  return 0;
}
``````
this code produce only: {"99":99} 
is this a bug or normaly?

bblanchon, 2017-12-12:
Hi @jliegner,

By default ArduinoJson doesn't copy strings, it just stores pointers.
In the program above, the same pointer is inserted 100 times; so, from ArduinoJson, standpoint it's the same key.

One way to fix this program is to call [`JsonBuffer::strdup()`](http://arduinojson.org/api/jsonbuffer/strdup/).
See demo here: https://wandbox.org/permlink/UpeAYp7Q9IIN9c4Q

The other way is to use an `std::string`:
See demo here: https://wandbox.org/permlink/Tp9FVkRirDUaukPC

In both cases, a copy of the string is made in the `JsonBuffer`.

See also:

* [FAQ: Using a loop, why are values all the same?](http://arduinojson.org/faq/using-a-loop-why-are-values-all-the-same/)
* [FAQ: JsonObject::set()](http://arduinojson.org/api/jsonobject/set/)

Regards,
Benoit

jliegner, 2017-12-12:
i understand, thank you.

bblanchon, 2017-12-13:
You're welcome.
Thanks for using ArduinoJson.

closed 2017-12-13 by bblanchon