bblanchon/ArduinoJson #638: Problem decoding PHP-generated content

boris-betc, 2017-12-14:
Hello,
Firstly, thanks for your usefull library :)

There is an issue that make me crazy for hours now.

If I want to parse a classic .json file (), no problem.
**Exemple : http://borislb.eu/bouchon.json**

If I want to parse a php-generated json content, I get the "JSON parsing failed!
" message (from the readReponseContent() function).
**Exemple : http://borislb.eu/arduinojson/index.php**

The content is exactly the same, I can't understand what does not work when it's a PHP generated content.

Thanks for your help !

bblanchon, 2017-12-15:
Hello @boris-betc,

My intuition tells me that the server uses [Chunked Transfer Encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding), but the client doesn't support it.
Try to switch to HTTP 1.0.

See:

* [FAQ: Why parsing fails?](https://arduinojson.org/faq/why-parsing-fails/)

Regards,
Benoit

boris-betc, 2017-12-20:
Hello @bblanchon ,
Thanks for your answer, and sorry for the late reply.
Thanks for your solution, it works :)

closed 2017-12-20 by boris-betc