bblanchon/ArduinoJson #639: Free JsonObject ?

timpur, 2017-12-14:
Do I need to free JsonObject and JsonAaray ? Since aren't they created dynamically ?

https://github.com/bblanchon/ArduinoJson/blob/b55e57a7cf6c0910186ed09cc3717a272982a522/src/ArduinoJson/JsonBufferImpl.hpp#L15

bblanchon, 2017-12-15:
Hi @timpur,

No, you don't need to delete `JsonArray`s and `JsonObject`s.
They are allocated inside the `JsonBuffer` so the memory is released when you destroy the `JsonBuffer`.

Regards,
Benoit

timpur, 2017-12-15:
I get that but won't the object it self use abit of memory ? Not just the content in the object ?

bblanchon, 2017-12-16:
Let's look at this statement again:

```c++
 JsonObject *ptr = new (this) JsonObject(this); 
```

This is not a regular call to the `new` operator, nor is it a `placement new`.
It is a call to an overloaded `new` operator in the base class [`JsonBufferAllocated`](https://github.com/bblanchon/ArduinoJson/blob/e53e4e3dd9b5571e37a1218e84dfd6bd2dc6d960/src/ArduinoJson/Data/JsonBufferAllocated.hpp#L14):

```c++
void *operator new(size_t n, JsonBuffer *jsonBuffer) throw() {
    if (!jsonBuffer) return NULL;
    return jsonBuffer->alloc(n);
}
```

So `ptr` in the first statement is set to the result of `jsonBuffer->alloc(n)`, which is an address inside the `JsonBuffer`.

timpur, 2017-12-16:
Oh damn my C++ is nothing compared to yours !!!!! Forgive me for having doubt.

So it really is just a pointer to the buffer. That's good to know and thank you for explaining this and I hope others find this use full also.

bblanchon, 2017-12-17:
You're welcome @timpur.

> *It is not the language that makes programs appear simple.
> It is the programmer that make the language appear simple!*
> -- Robert C Martin

Thank you for your interest in ArduinoJson.
Don't forget to add a star if you like this library ;-)

closed 2017-12-17 by bblanchon