bblanchon/ArduinoJson #640: Conan package

ZaMaZaN4iK, 2017-12-15:
Hello,
Do you know about [Conan](https://github.com/conan-io/conan)?
[Conan](http://docs.conan.io/en/latest/) is modern dependency manager for C++. And will be great if your library will be available via package manager for other developers.

[Here](https://github.com/bincrafters/conan-templates) you can find example, how you can create package for the library. 

If you have any questions, just ask :-)

bblanchon, 2017-12-16:
Hi @ZaMaZaN4iK,

Yes, I do know about Conan, and I even knew about Biicode before that; but, I never used them in any project.

I was wondering if some users would be interested in a Conan package; now, I know the answer :-)

One question though: does it makes sense to make a package for a single header library?

Regards,
Benoit

ZaMaZaN4iK, 2017-12-16:
Yes, making package even for single header library makes sense. For user doesn't matter how is written your library. User wants just press magic button and have library installed.

So i strongly recommend create package :-) And in your situation it will be very easy for you - just use example for header-only library: https://github.com/bincrafters/conan-templates/tree/header-only

bblanchon, 2017-12-18:
Hi @ZaMaZaN4iK,

I made the first step; I adapted the template to ArduinoJson:
https://github.com/bblanchon/conan-ArduinoJson

What's the next step?

ZaMaZaN4iK, 2017-12-18:
Next step: configure CI and test on different platforms. Also i suggest you visit cpplang.slack.com , channels #conan and #bincrafters . Here you will find a lot useful information. Also you can send your conan recipe to bincrafters repo.

ZaMaZaN4iK, 2017-12-18:
>I don't want to invest a lot of time on this because I believe very few users will use conan.
It's not true.

>Do you want to take care of that?
Yes, i can port your library to bincrafters repo.

bblanchon, 2017-12-18:
Thanks!
What do you need from me ?

ZaMaZaN4iK, 2017-12-18:
>What do you need from me ?

I think nothing for now :-) Will be great, if you will sometimes look into Conan recipe for your library :-)

bblanchon, 2017-12-27:
@ZaMaZaN4iK, any progress on the Conan package?

ZaMaZaN4iK, 2017-12-27:
https://github.com/bincrafters/conan-ArduinoJson/tree/testing/5.12.0

bblanchon, 2018-01-22:
Hi @ZaMaZaN4iK,

I see that this is still in "testing", is there anything we should do at this point?
Are you waiting for some action on my side?

Regards,
Benoit

bblanchon, 2018-01-31:
@danimtb pushed this package:
https://bintray.com/conan-community/conan/arduinojson:conan

closed 2018-01-31 by bblanchon

danimtb, 2018-01-31:
@ZaMaZaN4iK I didnt realize you were already developing an ArduinoJson package for conan. I apologize for that and would like to stand here I am willing to colaborate on future packages.

@bblanchon thanks for your feedback. I am actually a user of your library and I am currently trying to migrate some of the most popular arduino libraries to conan in order to perform crosscompilation for Arduino like devices.

Would you like to officially support the conan package for your library? It is very straight-forward as it is header only and not tied to any Arduino specifics (I am talking about dependencies here).

Having a Conan package will make your library even easier to install for consumers (see the[ example using Poco libraries](http://docs.conan.io/en/latest/getting_started.html#a-timer-using-poco-libraries)), just:

```
$ conan install arduinojson/5.13.0@conan/stable
```

Please [have a look at the repo](https://github.com/danimtb/conan-arduinojson) (most files are just CI config) and tell me what you think.

Thanks again!