bblanchon/ArduinoJson #641: Nested Json

mihir-joshi, 2017-12-16:
I tired various example for accepting a json response and storing them ... But couldn't succeed in any way. 
The Json response come as follows : 

{"sun_position":{"altitude":46.92,"azimuth":170.41}}

Any kind of help would be appreciated 

P.S :- New to Coding on Arduino..

beicnet, 2017-12-16:
Hey @mihir-joshi,

Can you try this example?
```
const size_t bufferSize = JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + 50;
DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "{\"sun_position\":{\"altitude\":46.92,\"azimuth\":170.41}}";

JsonObject& root = jsonBuffer.parseObject(json);

float sun_position_altitude = root["sun_position"]["altitude"]; // 46.92
float sun_position_azimuth = root["sun_position"]["azimuth"]; // 170.41
```
You need to replace` json` with your actual fetch code.

Regards,
Viktor

mihir-joshi, 2017-12-16:
Hey Viktor i tired ur code but sadly it didnt work am attaching the arduino code have a look 
In serial Monitor am getting the following error : Unexpected response: HTTP/1.1 400 Bad Request

The Url for the api is as follows www.yantree-mirasol.com/api/trackers/5a322035f7a29000041d347d/sun_position

Arduino Code,

`#include <ArduinoJson.h>
#include <Ethernet.h>
#include <SPI.h>

void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  while (!Serial) continue;

  // Initialize Ethernet library
  byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
  if (!Ethernet.begin(mac)) {
    Serial.println("Failed to configure Ethernet");
    return;
  }
  delay(1000);

  Serial.println(F("Connecting..."));

  // Connect to HTTP server
  EthernetClient client;
  client.setTimeout(10000);
  if (!client.connect("https://www.yantree-mirasol.herokuapp.com", 80)) {
    Serial.println("Connection failed");
    return;
  }

  Serial.println("Connected!");

  // Send HTTP request
  client.println("GET https://www.yantree-mirasol.com/api/trackers/5a322035f7a29000041d347d/sun_position");
  client.println("HTTP/1.1");
  client.println("Host: yantree-mirasol.herokuapp.com");
  client.println("Accept: application/vnd.mirasol.v1");
  client.println("Connection: close");
  if (client.println() == 0) {
    Serial.println("Failed to send request");
    return;
  }

  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print("Unexpected response: ");
    Serial.println(status);
    return;
  }

  // Skip HTTP headers
  char endOfHeaders[] = "\r\n\r\n";
  if (!client.find(endOfHeaders)) {
    Serial.println("Invalid response");
    return;
  }

  // Allocate JsonBuffer
  // Use arduinojson.org/assistant to compute the capacity.
 const size_t bufferSize = JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(2) + 50;
 DynamicJsonBuffer jsonBuffer(bufferSize);

const char* json = "{\"sun_position\":{\"altitude\":46.92,\"azimuth\":170.41}}";

JsonObject& root = jsonBuffer.parseObject(json);

  // Parse JSON object
  //JsonObject& root = jsonBuffer.parseObject(client);
  if (!root.success()) {
    Serial.println(F("Parsing failed!"));
    return;
  }
float sun_position_altitude = root["sun_position"]["altitude"]; // 46.92
float sun_position_azimuth = root["sun_position"]["azimuth"]; // 170.41
Serial.println(sun_position_altitude);
Serial.println(sun_position_azimuth);
  // Extract values
//  Serial.println(F("Response:"));
//  Serial.println(root["sensor"].as<char*>());
//  Serial.println(root["time"].as<char*>());
//  Serial.println(root["data"][0].as<char*>());
//  Serial.println(root["data"][1].as<char*>());

  // Disconnect
  client.stop();
}

void loop() {
  // not used in this example
}`

bblanchon, 2017-12-17:
Hi @mihir-joshi,

The syntax of the HTTP request in incorrect:

```c++
client.println("GET http://yantree-mirasol.herokuapp.com/api/trackers/5a322035f7a29000041d347d/sun_position");
client.println("HTTP/1.1");
```

must be replaced by:

```c++
client.println("GET /api/trackers/5a322035f7a29000041d347d/sun_position HTTP/1.0");
```

Regards,
Benoit

danielsippel, 2017-12-18:
And furthermore replace 

` if (!client.connect("https://www.yantree-mirasol.herokuapp.com", 80)) {`

with

` if (!client.connect("http://www.yantree-mirasol.herokuapp.com", 80)) {`

I'm currently searching for a solution to get the request working via https/ssl :-)

mihir-joshi, 2017-12-18:
Hi @bblanchon ,
I tried made the changes according to your suggestion but still it did not work... Am getting the same error the code in which api is written is in Ruby on Rails

mihir-joshi, 2017-12-18:
UPDATE:- Now the error code changed and it went from HTTP/1.1 400 Bad Request to HTTP/1.1 500 Internal Server Error 

Note that the content Altitude and Azimuth in the api reponse is in float

bblanchon, 2017-12-18:
Hi @mihir-joshi,

I'm sorry but this problem is not related to ArduinoJson. It's already a lot of work to answer questions about ArduinoJson alone, we cannot be a help desk for all programming questions. I invite you to turn to Stack Overflow or Arduino forum.

Thank you for your understanding.

Regards,
Benoit

closed 2017-12-18 by bblanchon