bblanchon/ArduinoJson #642: JSON with top level array

osresearch, 2017-12-16:
JSONlint validates input with a top-level array object: https://jsonlint.com/?json=[0,%22foo%22,2]

However, attempting to parse that with ArduinoJson results in an error:

```
  StaticJsonBuffer<200> jsonBuffer;
  char json[] = "[0,\"foo\",2]";
  JsonObject& root = jsonBuffer.parseObject(json);
  if (!root.success()) {
    std::cerr << "parseObject() failed" << std::endl;
    return 1;
  }
```

Additionally, trying to retrieve a top-level array item results in a compile-time error:

```
  const char * foo = root[1];
  std::cout << foo << std::endl;
```

```
In file included from prog.cc:9:0:
ArduinoJson.h: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator ArduinoJson::JsonObject::findKey(TStringRef) [with TStringRef = const int&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator = ArduinoJson::Internals::ListIterator<ArduinoJson::JsonPair>]':
ArduinoJson.h:2520:66:   required from 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::const_iterator ArduinoJson::JsonObject::findKey(TStringRef) const [with TStringRef = const int&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::const_iterator = ArduinoJson::Internals::ListConstIterator<ArduinoJson::JsonPair>]'
ArduinoJson.h:2525:44:   required from 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson::JsonObject::get_impl(TStringRef) const [with TStringRef = const int&; TValue = const char*; typename ArduinoJson::Internals::JsonVariantAs<TValue>::type = const char*]'
ArduinoJson.h:2447:44:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type ArduinoJson::JsonObject::get(const TString&) const [with TValue = const char*; TString = int; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type = const char*]'
ArduinoJson.h:2965:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = const char*; TStringRef = const int&; typename ArduinoJson::Internals::JsonVariantAs<T>::type = const char*]'
ArduinoJson.h:257:35:   required from 'ArduinoJson::JsonVariantCasts<TImpl>::operator T() const [with T = const char*; TImpl = ArduinoJson::JsonObjectSubscript<const int&>]'
prog.cc:21:28:   required from here
ArduinoJson.h:2514:54: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'
       if (Internals::StringTraits<TStringRef>::equals(key, it->key)) break;
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
```

osresearch, 2017-12-16:
I figured out that I could parse it with `jsonBuffer.parseArray()`, although it is unexpected that it doesn't work with `parseObject()`.

bblanchon, 2017-12-17:
Hi @osresearch,

I'm sorry for the confusion:

* [JsonBuffer::parseArray()](https://arduinojson.org/api/jsonbuffer/parsearray/) deserializes a JSON array
* [JsonBuffer::parseObject()](https://arduinojson.org/api/jsonbuffer/parseobject/) deserializes a JSON object.

That sounds clear to me; what was your understanding?

Regards,
Benoit

osresearch, 2017-12-17:
The "Object" versus "Array" distinction was not clear to me -- I had assumed that the `parseArray()` method was the generic "turn this string into a JSON object", rather than requiring the caller to know something about the input.

bblanchon, 2017-12-18:
`parseObject()` and `parseArray()` mainly differ by their return type: `JsonObject&` or `JsonArray&`.
You often need to know the type at compile-time because the operations available are not the same.

There is a third function, [JsonBuffer::parse()](https://arduinojson.org/api/jsonbuffer/parse/), that works with both arrays and object, but it returns a `JsonVariant`.

closed 2017-12-22 by bblanchon