bblanchon/ArduinoJson #643: Json parsing

mihir-joshi, 2017-12-19:
I have created an API in ruby on rails and it send the following json response:

{"sun_position":{"altitude":"32.46","azimuth":"223.93"}}
Am able to successfully hit API and store the response. But when I try to parse it and store data it gives me the error. I have used Arduino JSON library

The Code for parsing the JSON is as follows:

bool parseSunPosData(char* content, SunPosData* sunPos) {
  StaticJsonBuffer<JSONBUFFER_SIZE> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(content);
  Serial.println("In parse fuc");
  Serial.println(content);
 // Serial.println(root);
  Serial.println("==============");

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
    return false;
  }
and the output on serial monitor is as follows:

In parse fuc 

38
{"sun_position":{"altitude":"32.46","azimuth":"223.93"}} 
0

==============

JSON parsing failed! Disconnect
My guess is it cant parse the data due to the numbers 38 and zero printing ... I don't know what that is and what it is

P.S:- New to Arduino coding

mihir-joshi, 2017-12-19:
@bblanchon
I got the solution i changes the HTTP from 1.1 to 1.0 and it worked like a charm

bblanchon, 2017-12-19:
Cool 👍 

Remark that the answer was in the first lines of [Why parsing fails?](http://arduinojson.org/faq/why-parsing-fails/)

closed 2017-12-19 by bblanchon