bblanchon/ArduinoJson #644: Parsing succes but no data after parsing

Disane87, 2017-12-22:
Hello, thank you for this great lib! 👍 

I'm just implementing it into my controller to register the controller at my gateway, so I know which devices in my house are active and sending data. Before that, I have to check if the device is already registered at my gateway:

```c++
bool checkRegistration(){
  // http://api.home.disane.de/index.php/Devices?columns=ID&filter%5B%5D=MAC%2Ceq%2C30%3AAE%3AA4%3A47%3A74%3AC0
  String URL = String(API_BASE)+"/Devices/?columns=ID&filter=MAC,eq,"+urlencode(WiFi.macAddress())+"&transform=1&page=1,1";
  Serial.println("[checkRegistration]: Calling API: "+URL);
  api.begin(URL);
  int httpCode = api.GET();

  if(httpCode) {
      Serial.println("[checkRegistration]: Got HTTP: "+String(httpCode));
      if(httpCode == 200) {
        Serial.println("[checkRegistration]: Return "+api.getString());
        JsonObject& ret = jsonBuffer.parseObject(api.getString());
        if (!ret.success()) {
          DeviceID = ret["Devices"][0]["ID"];
          ret.prettyPrintTo(Serial);
          Serial.println("[checkRegistration]: Got Device-ID: "+DeviceID);
          display_text("Registration done");
          api.end();
          if(DeviceID != 0){
            return true;
          }else{
            return false;
          }
        }else{
          Serial.println("[checkRegistration]: Could not parse JSON: ");
          return false;
        }
        
        
      }
    }else{
      Serial.print("[registerDevice]: Failed!");
      display_text("checkRegistration failed");
      api.end();
      return false;
    }
    
}
```

The serial outout:

> [registerDevice]: Registering Device
[checkRegistration]: Calling API: http://api.home.disane.de/Devices/?columns=ID&filter=MAC,eq,30%3AAE%3AA4%3A47%3A74%3AC0&transform=1&page=1,1
[checkRegistration]: Got HTTP: 200
[checkRegistration]: Return {"Devices":[{"ID":14},{"ID":15}]}
{}[checkRegistration]: Got Device-ID:
[registerDevice]: Calling API: http://api.home.disane.de/Devices
[registerDevice]: Begin call: http://api.home.disane.de/Devices
[registerDevice]: got response: 200
[registerDevice]: Device-ID: 16

The Json itself:

> {"Devices":[{"ID":14}]}

I plemented my solution according your assitant. But I don't get any data from "ret["Devices"][0]["ID"]". I really dont get why. The JSON is valid and I used both, parseArray and parseObject, but nothing works.

Can you help me?

bblanchon, 2017-12-25:
Hi @Disane87,

Please apologize for the long wait.

If seems that the following statement is wrong:

    if (!ret.success())

I think you should remove the negation.

Regards,
Benoit

Disane87, 2017-12-25:
Oh well, sorry! Merry xmas by the way! :) 
Doh! That was my fault 😐

Have some good holidays!

Disane87, 2017-12-25:
Just check it, it works, but there is no output. The JSON is parsed successfull, but there is no output with 
``
ret["Devices"][0]["ID"]
``

Disane87, 2017-12-25:
Now the JSON parsing is not successfull, but why? The JSON should be valid and your assistant drop out the correct query. :-(

bblanchon, 2017-12-26:
I'm @Disane87, what improvement are you talking about?

closed 2018-01-02 by bblanchon