bblanchon/ArduinoJson #645: I'm going crazy.... Parse Json object in array

Tylerdurden73, 2017-12-24:
Hi,  
Could any one hell me please... 
I want to parse three follow data file... 
[{“name“:"test 1","something":"123"},{“name“:"test 2","something":"1234"}]
The file I become from web. File transfer is OK. How can I parse this data? 
I hope someone can help me.  I use a esp 8266 esp12 with arduino ide. 
I want to store the name value in a string. 

Thanks a lot
Frank

beicnet, 2017-12-24:
Hey Frank,

The issues is that it's not a valid Json structure as I see:

```
[
  {
    “name“: "test 1",
    "something": "123"
  },
  {
    “name“: "test 2",
    "something": "1234"
  }
]
```

Regards,
Viktor

Tylerdurden73, 2017-12-24:
Thanks for quick answere. Vitor.
Where is the different between my and yours Json?  
And how can I find out how much objects are in the array.... Thank you so much and merry Christmas

beicnet, 2017-12-25:
The quotation is wrong in your Json object example, see... “ against " do you see the difference?

What do you mean, object size or count?

If you use it like this, you don't need to worry about the size:
```
JsonData = ""; // Your Json input data to be parsed
DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(JsonData);
```

and Happy Christmas to you too!! :wink:

bblanchon, 2017-12-25:
Hi @Tylerdurden73,

I agree with @beicnet, the quotation marks around `name` are not the right character.

To know the number of elements in an array, call [`JsonArray::size()`](https://arduinojson.org/api/jsonarray/size/), but to list all elements in an array, use [`JsonArray::begin() / JsonArray::end()`](https://arduinojson.org/api/jsonarray/begin_end/).

Regards,
Benoit

Tylerdurden73, 2017-12-26:
Oh OK...  I see. I wrote it only different here.  It works fine now... And I slept very good 2 nights😂 thank s to all.

closed 2017-12-26 by Tylerdurden73