bblanchon/ArduinoJson #647: root.printTo(Serial) works fine but root.printTo(client) on my Arduino MKRGSM1400 doesn't

johannstieger, 2017-12-26:
Hi Benoît,

I do have this Arduino-Code on my MKRGSM1400:
StaticJsonBuffer<500> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
root["Gateway"] = 1;
root["IMEI"] = modemTest.getIMEI();
root["CurrentCarrier"] = scannerNetworks.getCurrentCarrier();
root["GSMSignalStrenght"] = scannerNetworks.getSignalStrength();
...
...
and when I do a root.printTo(Serial); I get this:
{"Gateway":1,"IMEI":"357520074425659","CurrentCarrier":"HoT","GSMSignalStrenght":"2","GatewayIP":178977430,"GatewayTime":1514289175}
Serial.println(root.measureLength()) tells me a lenght of 132, which is right.

But when I do a root.printTo(client); to transmit the JSON Object to a nginx web-server over the GSM Network I do receive this string on my web-Server (I received it with file_get_contents("php://input")):
{}"{Gateway"y:"1,Ü",IMEI"I:"":357520074425659"9,"",CurrentCarrier"r:"":HoT"T,"",GSMSignalStrenght"t:"":2"2,"",GatewayIP"P:"17897743

The lenght is correct but the content is wrong.

After that I used tcpdump on my web-server to check where there could be the problem. 
When I make the post-request with curl the tcpdump looks like this:
POST /Fallenmelder/GW.php HTTP/1.1
User-Agent: curl/7.35.0
Host: johannstieger.myds.me
Accept: */*
Content-Type: application/json
Content-Length: 133

{"Gateway":1,"IMEI":"357520074425659","CurrentCarrier":"HoT","GSMSignalStrenght":"5","GatewayIP":1684314475,"GatewayTime":1514129328}

But when I make the same from my arduino I get this dump:
POST /Fallenmelder/GW.php HTTP/1.1
Content-Type: application/json
Content-Length: 133
Connection: close
Host: johannstieger.myds.me

{}"{Gateway"y:"1,..",IMEI"I:"":35752
12:31:09.167987 IP 192.168.186.254.http > 46.125.249.58.64802: Flags [.], ack 176, win 15544, length 0
E..(..@.@........}.:.P."+}z.....P.<..y..
12:31:09.268404 IP 46.125.249.58.64802 > 192.168.186.254.http: Flags [P.], seq 176:225, ack 1, win 64240, length 49: HTTP
E..Y..@....:.}.:.....".P....+}z.P.......0074425659"9,"",CurrentCarrier"r:"":HoT"T,"",GSMS
12:31:09.268440 IP 192.168.186.254.http > 46.125.249.58.64802: Flags [.], ack 225, win 15544, length 0
E..(..@.@........}.:.P."+}z....NP.<..y..
12:31:09.368419 IP 46.125.249.58.64802 > 192.168.186.254.http: Flags [P.], seq 225:286, ack 1, win 64240, length 61: HTTP
E..e..@....-.}.:.....".P...N+}z.P.......ignalStrenght"t:"":5"5,"",GatewayIP"P:"1684314975,..",Gateway
12:31:09.368457 IP 192.168.186.254.http > 46.125.249.58.64802: Flags [.], ack 286, win 15544, length 0
E..(..@.@........}.:.P."+}z.....P.<..y..
12:31:09.369440 IP 192.168.186.254.http > 46.125.249.58.64802: Flags [P.], seq 1:187, ack 286, win 15544, length 186: HTTP: HTTP/1.1 200 OK
E.....@.@........}.:.P."+}z.....P.<..3..HTTP/1.1 200 OK
Server: nginx
Date: Tue, 26 Dec 2017 11:31:09 GMT
Content-Type: text/html; charset=UTF-8
Transfer-Encoding: chunked
Connection: close
Vary: Accept-Encoding

So it seems the problem is on the arduino but I can not figure out if there is a problem on ArduinoJson or the client-library of GSMSSLClient or GSMClient (I tested it with and without SSL).

I hope you can help me to solve this problem.

Many thanks, Johann Stieger

bblanchon, 2017-12-26:
Hi Johann,

Can you try to print the JSON document to a temporary buffer?
Then use this buffer to send to the GSM client.

Regards,
Benoit

johannstieger, 2017-12-26:
Hi Benoît,
thank you very much for your support - these lines solved my problem:
   String JSBuffer;
   root.printTo(JSBuffer);
   client.println(JSBuffer);
As a little thank to you I bought your new book from Leanpub.
Best regards, Johann

bblanchon, 2017-12-27:
Hi Johann,

Thank you very much for buying the book, I hope you'll enjoy it.

I'm glad you found a solution for the GSMClient, but the original implementation should have worked.

Can you give me a link to the source code of GSMClient?
I thought it was in [github.com/arduino/ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd), but I can't find it.

Regards,
Benoit

closed 2018-01-02 by bblanchon