bblanchon/ArduinoJson #648: Q:  represent ArduinoJSON char in human readable format

aboulfad, 2017-12-26:
Hello,
I am sorry for creating this question, but I can't figure it out ... (apologies if this is NOT related to ArduinoJSON), but is it possible at all ? Mille millions de mille sabords ;-) 
* Description: I would like to re-write the following in an human readable format so it compiles.
```
-One line version: char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

-Human readable Multi-line version: 
char json[] = {
"sensor":"gps",
"time":1351824120,
"data":[48.756080,2.302038]
}";
```
* target platform: ESP8266
* compiler model and version: Arduino IDE 1.8.5
* compiler output: exit status 1 expected '}' before ':' token

bblanchon, 2017-12-27:
Hi @aboulfad,

If you use C++11, you can write:

```c++
char json[] = R"(
 {
   "sensor": "gps",
   "time": 1351824120,
   "data": [48.756080,2.302038]
 })";
```

See demo: https://wandbox.org/permlink/PP6DqjI5Mtio2Uyc

Caution: the string is much bigger as it now includes the spaces and line breaks.

You can also split the string literal by ending each line with a quote mark:

```c++
char json[] = 
 "{"
   "\"sensor\": \"gps\","
   "\"time\": 1351824120,"
   "\"data\": [48.756080,2.302038]"
 "}";
```

But, you lose the "human readable" aspect.

**As you said, this question is not related to ArduinoJson.
Please, have mercy on library developers and don't use GitHub as a help desk.
Instead, go to [Stack Overflow](https://stackoverflow.com/) or [Arduino Forum](http://forum.arduino.cc/).**

Regards,
Benoit

closed 2017-12-27 by bblanchon

aboulfad, 2017-12-27:
Merci pour ta réponse,  j’ai acheté une copie de ton livre pour te remercier :-)

bblanchon, 2017-12-28:
Merci, si tout le monde faisait comme toi, je serais riche 💰 
J'espère que tu apprécieras le livre.

aboulfad, 2017-12-28:
Tu est déjà riche sans que tu le sache . La richesse dans la vie, c’est aide les autres sans rien demande. Ton ArduinoJson et ton support est un example 😄