bblanchon/ArduinoJson #649: jsonArray.add JsonObject& causes crash

AlexBokhankovich, 2017-12-27:
Device:ESP8266;
Arduino:latest,
ArduinoJson:latest

I have the following code to serialize my std::vector<Mode*> modes to json. Each Mode has std::vector<Parameter*> parameters.
Parameter has function toJson that converts in to JsonObject&.
In case if I try to add nested parameter my device crases on printing jsonObject to string.
```
                String json;
		auto modesCount = modes.size();
		const int capacity = modesCount * JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(modesCount) + modesCount * JSON_OBJECT_SIZE(modesCount) + 8 * JSON_OBJECT_SIZE(5) +700;
		DynamicJsonBuffer jsonBuffer(capacity);
		auto& root = jsonBuffer.createArray();
		for (Mode* mode : modes)
		{
			auto& element = root.createNestedObject();
			element["id"] = mode->Id;
			element["name"] = mode->Name;
			element["isActive"] = mode->Id == currentMode->Id;

			auto& parameters = element.createNestedArray("parameters");
			for (Parameter* parameter : mode->parameters)
			{
				//"min":0, "max" : 255, "type" : 1, "value" : 125, "id" : "brightness", "name" : "Brightness"
				auto& val = parameter->toJson();
				parameters.add(val);
			}
			
		}

		root.printTo(json);
```

What could be the reason of such behavior and how this can be fixed or workarounded?

AlexBokhankovich, 2017-12-27:
I have a workaround for this. Iheve added Type field to Parameter, so I can identify which type parameter I have and then I cast parameter to concrete type where I can get any type specific fields.
```
auto& parameters = element.createNestedArray("parameters");
			for (Parameter* parameter : mode->parameters)
			{
				auto& param = parameters.createNestedObject();
//				//"min":0, "max" : 255, "type" : 1, "value" : 125, "id" : "brightness", "name" : "Brightness"
				param["id"] = parameter->id;
				param["name"] = parameter->name;
				param["type"] = static_cast<int>(parameter->type);
				param["value"] = 5;

				switch (parameter->Type) { 
				case ParameterType::Number:
					{
					auto num = static_cast<NumberParameter*>(parameter);
					}
					break;
				case ParameterType::Range:
				{
					auto num = static_cast<RangeParameter*>(parameter);
					param["max"] = num->max;
					param["min"] = num->min;
				}break;
				case ParameterType::Boolean:
					{
					auto num = static_cast<BooleanParameter*>(parameter);
					param["value"] = true;
					}
					break;
				case ParameterType::Color: {
					auto num = static_cast<ColorParameter*>(parameter);
					param["value"] = "red";
				}
					break;
				default: ;
				}
			}
```

Is there a better workaround?

bblanchon, 2017-12-27:
Hi @AlexBokhankovich,

This line looks very suspicious:

```c++
auto& val = parameter->toJson();
```

I assume this returns a `JsonObject&`, but where is the `JsonBuffer`?

Regards,
Benoit

AlexBokhankovich, 2017-12-27:
Hi @bblanchon,
Thank you for your reply. `JsonBuffer` is inside of Parameter class' toJson function. I got your poitn, it looks like several `JsonBuffer`s  can not work together. Here is My toJson function listing

```
JsonObject& ColorParameter::toJson()
{
	// https://arduinojson.org/assistant/

	const int capacity = JSON_OBJECT_SIZE(6) + 56;
	DynamicJsonBuffer jsonBuffer(capacity);

	auto& root = jsonBuffer.createObject();
	//{"min":0, "max" : 255, "type" : 1, "value" : 1, "id" : "brightness", "name" : "Brightness"}
	root["max"] = 255;
	root["type"] = 1;
	root["value"] = FastLED.getBrightness();
	root["id"] = id;
	root["name"] = name;

	return root;
}
```

Diaoul, 2017-12-27:
Total noob in C++ however I had this issue and ended up passing the jsonBuffer as argument to that function.
To my understanding the buffer gets garbage collected as soon as you exit the function, and the returned JsonObject leads to freed memory and crashes. Sometimes it works though, I guess the garbage collection isn't deterministic, doesn't necessarily happen when you exit the function.

Honestly I'm way over my head here so I will not attempt further to describe what is happening :smile:

bblanchon, 2017-12-28:
Hi,

@AlexBokhankovich, the problem is that `auto& root` is a reference to an object inside the `JsonBuffer` which is destructed as soon as the function exits.

The suggestion of @Diaoul can work, but my prefered solution is to pass the `JsonObject&` to the function:

```c++
void ColorParameter::toJson(JsonObject& obj)
{
	//{"min":0, "max" : 255, "type" : 1, "value" : 1, "id" : "brightness", "name" : "Brightness"}
	obj["max"] = 255;
	obj["type"] = 1;
	obj["value"] = FastLED.getBrightness();
	obj["id"] = id;
	obj["name"] = name;
}
```

```c++
for (Parameter* parameter : mode->parameters)
{
	parameter->toJson(parameters.createNestedObject());
}
```

It is the solution I used in the case study "Configuration in SPIFFS" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Regards,
Benoit

closed 2018-01-02 by bblanchon