bblanchon/ArduinoJson #651: JsonVariant::success() returns true for NULL values

timpur, 2017-12-29:
How can i tell if a key has a value or not? `JsonVariant::success()` doesnt seem to always work.
`JsonVariant::success()` can be true when `value.as<const char*>() == nullptr` is also true, which means the value is NULL. Am i doing something wrong?

```
char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038],\"test\":null}";

if(root["test"].success()){
  std::cout << "test has a value" << std::endl;
    if(root["test"].as<const char*>() == nullptr){
      std::cout << "test value is null" << std::endl;
    }
}
```

bblanchon, 2017-12-30:
Hi @timpur,

[JsonVariant::success()](https://arduinojson.org/api/jsonvariant/success/) tells whether the variant is defined or not.

`null` is a valid value in JSON; so, if a variant contains `null`, `success()` logically returns `true`.

Have a look [JsonVariant::operator|](https://arduinojson.org/api/jsonvariant/or/) which doesn't consider `null` as a valid value.
I use this technique in the case study "Configuration in SPIFFS" in [Mastering ArduinoJson](https://arduinojson.org/book/).

Regards,
Benoit

timpur, 2017-12-30:
Love it 
Thank you !!!

bblanchon, 2017-12-31:
You're welcome.
Thank you for using ArduinoJson.

closed 2017-12-31 by bblanchon

timpur, 2018-01-16:
@bblanchon, is it possible to get in contact with you. Like to ask you for advice for a new Arduino lib idea / contrute if you like to.

timpur, 2018-02-11:
@bblanchon ? Any chance to get in contact with you ?

bblanchon, 2018-02-12:
@timpur On January 16th, I sent you an email to the address that you use in your commits on GitHub; apparently, you didn't receive it.

You can also contact me on [LinkedIn](https://www.linkedin.com/in/benoit-blanchon/), or [Twitter](https://twitter.com/BenoitBlanchon).

timpur, 2018-02-12:
Awks, spam filter.... Thanks.