bblanchon/ArduinoJson #652: Add JSON_NODE_SIZE (in addition to JSON_OBJECT_SIZE and JSON_ARRAY_SIZE)

Diaoul, 2017-12-29:
I'm building a JSON-based FastLED effect library. Each Effect is responsible for a portion of the JSON under the `effect` `JsonObject`:

Inheritance is as follow: BaseEffect > PaletteEffect > TwinkleEffect
```
{
   "brightness": 100,
   "effect": {               // handled by the main class
      "name": "twinkle",     // handled by the BaseEffect class
      "palette": "rainbow",  // handled by the PaletteEffect class
      "density": 255,        // handled by the TwinkleEffect class
      [...]                  // handled by the TwinkleEffect class
   }
}
```
At some point I need to report the total size of the Json to allocate my buffer with proper length. My idea is that each class report what they use.

As `2 * JSON_OBJECT_SIZE(1) > JSON_OBJECT_SIZE(2)`, I cannot use that.
I cannot store objectSize and arraySize and compute the `JSON_OBJECT_SIZE(objectSize) + JSON_ARRAY_SIZE(arraySize)` because that would mean the `effect` `JsonObject` could not contain other nested nodes.

So I ended up with this:
```c
#define JSON_NODE_SIZE \
  (sizeof(JsonObject::node_type))
```
Which IMO should be included in ArduinoJson.

Moreover, it's easy to count (as an alternative):

* Number of JsonObject = `{` count
* Number of JsonArray = `[` count
* Number of Node = number of keys | array items

What do you think?

bblanchon, 2017-12-30:
Hi @Diaoul,

It looks like you are overthinking the size of the buffer.

Here is my opinion: if your program computes `JsonBuffer`'s size at **run-time**, you're trying to be too clever.

Remember why we specify the size of the buffer: 

1. to allocate on the stack (only for `StaticJsonBuffer`)
2. to reduce heap fragmentation (only for `DynamicJsonBuffer`)

You should not be looking for the exact memory consumption in each possible configuration.
Instead, you should pick one size that is big enough to support all valid configurations.

If the size depends on the configuration, use the worst case scenario, and use the [assistant](https://arduinojson.org/assistant/) to compute the size.
If you want an elastic size, let the `DynamicJsonBuffer` grow; the simplification of the program is worth the small waste in memory.

Regards,
Benoit

Diaoul, 2017-12-31:
This is exactly what I am doing: computing the maximum buffer size on which I can fit all JSON possibilities. I'm still trying to make this a compile-time fixed (yet computed) value.

The assistant cannot help as the JSON will depend on the effects you choose and this is customizable.

Memory is an issue on Arduino, I'm trying to save a few bytes wherever I can. I'm already working with FastLED that is very efficient however stills needs memory to store individual LED colors, that can quickly grow.

bblanchon, 2017-12-31:
OK, thanks for the clarification.
However, I don't understand how `JSON_NODE_SIZE` solves the problem.
Can you share some code?

bblanchon, 2018-01-10:
@Diaoul, did you see my answer? Can you share some code?

Diaoul, 2018-01-15:
Didn't have time to work on making the code publishable yet. I'll update
once this is done.

bblanchon, 2018-01-26:
@Diaoul, I'm closing this issue for the moment, please reopen when you get more information.
Thank you for your understanding.

Diaoul, 2018-01-26:
Sure! No problem. I'll let you know when this is ready.

Diaoul, 2018-02-04:
[Here](https://github.com/Diaoul/LEDEffect) it is.
```json
{
   "state": "OFF",
   "brightness": 1,
   "brightness_rate": 8,
   "fps": 30,
   "effect": {
      "name": "applause",
      "blend": false,
      "palette": "ocean",
      "fade_rate": 32
   }
}
```
The `effect` part is [dynamic](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L24-L29) and thus the maximum size of the JSON depends on the [selected effects](https://github.com/Diaoul/LEDEffect/blob/master/examples/esp8266/src/main.cpp#L130-L137). As this is a library I didn't want to enforce an arbitrary size or delegate the buffer management to userland.

bblanchon, 2018-02-05:
Hi @Diaoul,

I'm pretty sure the computed capacity is wrong.
Did you compare it with [`JsonBuffer::size()`](https://arduinojson.org/api/jsonbuffer/size/?utm_source=github&utm_medium=issues)?

Again, I don't see the point, as I said:

> If your program computes JsonBuffer's size at run-time, you're trying to be too clever.
>
> Let the DynamicJsonBuffer grow; the simplification of the program is worth the small waste in memory.

Regards,
Benoit

PS: that's a very cool project, do you want to have it featured in [ArduinoJson's News](https://arduinojson.org/news/?utm_source=github&utm_medium=issues)?

Diaoul, 2018-02-05:
[It](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L109-L112) [is](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L134-L137) [compared](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L147-L150) [everywhere](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L160-L163) and the size is correct.

I compute it anyway so users of the library can know exactly after a run the size of the `StaticJsonBuffer` they should use if they choose to handle the [deserialization ](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L45)and [serialization ](https://github.com/Diaoul/LEDEffect/blob/master/src/LEDEffect/LEDEffect.hpp#L117)themselves. So I thought I might as well use it...

bblanchon, 2018-02-06:
You have my opinion; it is your right to ignore it.

Anyway, I don't think `JSON_NODE_SIZE` would help the majority of users; instead, it will only add confusion.
Moreover, we would have to add `JSON_OBJECT_NODE_SIZE` and `JSON_ARRAY_NODE_SIZE` because the values are different.

Diaoul, 2018-02-06:
I don't want to ignore it I just want to understand it. Now that I've been through the trouble of computing it I'm just looking for a good reason to drop it altogether.

Not so long ago the docs basically said that `DynamicJsonBuffer` should be avoided at all cost and that `StaticJsonBuffer` shouldn't be global. Now it's unclear whether there is a preferred kind of buffer and one can clear the `StaticJsonBuffer`.
Has something changed? Is there a risk using the `DynamicJsonBuffer`? How will it grow by itself, 100 by 100?

I initially wanted to go for the `Static` kind but I haven't enough knowledge of C++ to make it work with various kind of effects. Maybe it's possible with templates but I don't even know.

Do you think allowing the users of the lib to use their own buffer is a waste?

bblanchon, 2018-02-07:
That's not what I meant. Let me explain.

The goal of software design is to manage the complexity.
When you solve a simple problem with a complex solution, you add complexity.
In your case, the problem is "I don't know the size of the JsonBuffer".
We know there is a simple solution to this problem, why choose a complex solution instead?
If I were you, I would choose the sub-optimal solution, which is to let the `DynamicJsonBuffer` grow; thereby reducing the complexity and improving the maintainability.

On the other hand, if the goal was to use a `StaticJsonBuffer`, the problem would have been different.
Instead of "I don't know the size of the JsonBuffer", it's now "I need to tell the compiler the size of the JsonBuffer".
This is a *complex* problem that requires a *complex* solution.
But is it worth it? What is your target platform? What is the expected size?
As soon as you target an ESP8266 or bigger, it's better to bet on `DynamicJsonBuffer` anyway, because the stack is limited to 4KB.

It's not that `DynamicJsonBuffer` should be avoided at all cost; it's just that you should prefer a `StaticJsonBuffer` if possible (maybe the doc needs to be updated).
Moreover, if you program uses `String`, it already relies on dynamic allocation and elastic size, so using a `DynamicJsonBuffer` won't impact the performance.

Lastly, if your users want to perform the JSON serialization/deserialization by themselves, they can simply use the assistant or look at the `JsonBuffer::size()` after the first run.
Again: simple problem, simple solution.

closed 2018-03-06 by bblanchon