bblanchon/ArduinoJson #653: Creating JsonObject from large json string.

Linkaan, 2018-01-01:
First of all, thanks for the awesome library. It seems really useful for parsing JSON not only using an Arduino but probably could be used with a Raspberry Pi aswell.

I'm trying to run a neural network on an Arduino UNO with the ATmega328-PU chip. I have created an environment to train the neural network and then exported the serialized neural network to json. My hope is to be able to run the neural network on the arduino uno.

The problem I have right now is that the JSON is too large for the arduino to handle with only 2KB of dynamic memory. My first attempt was simply to use https://arduinojson.org/assistant/ to deserialize the JSON into a jsonbuffer (awesome tool by the way), however that was not successful because it used 277% of the memory available.

What I would like to know is if there is any way to store a static json buffer into flash memory directly as there is plenty of memory available there. In other words is there any way to parse the json at compile time?

**MVCE:**
Here is the program that was generated using the assistant tool with a few modifications to print out some debug information.
```
#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  while (!Serial) continue;
  const size_t bufferSize = 6*JSON_ARRAY_SIZE(0) + 2*JSON_ARRAY_SIZE(1) + 5*JSON_ARRAY_SIZE(2) + 3*JSON_ARRAY_SIZE(3) + JSON_ARRAY_SIZE(4) + JSON_ARRAY_SIZE(5) + 2*JSON_ARRAY_SIZE(8) + JSON_ARRAY_SIZE(10) + JSON_OBJECT_SIZE(2) + 46*JSON_OBJECT_SIZE(4) + 10*JSON_OBJECT_SIZE(9) + 5030;
  DynamicJsonBuffer jsonBuffer(bufferSize);

  char json[] = "{\"neurons\":[{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":0,\"toNeuron\":6,\"weight\":0.3998056650161743,\"isRecurrent\":false},{\"fromNeuron\":0,\"toNeuron\":9,\"weight\":0.5525347590446472,\"isRecurrent\":false},{\"fromNeuron\":0,\"toNeuron\":39,\"weight\":-1.4688773155212402,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":0.7231108546257019,\"type\":0,\"neuronID\":0,\"activationResponse\":1,\"splitX\":0.2857142984867096,\"splitY\":0},{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":1,\"toNeuron\":6,\"weight\":1.3139477968215942,\"isRecurrent\":false},{\"fromNeuron\":1,\"toNeuron\":7,\"weight\":-1.841086983680725,\"isRecurrent\":false},{\"fromNeuron\":1,\"toNeuron\":9,\"weight\":0.33449316024780273,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":1.1387290954589844,\"type\":0,\"neuronID\":1,\"activationResponse\":1.0778632164001465,\"splitX\":0.4285714626312256,\"splitY\":0},{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":2,\"toNeuron\":6,\"weight\":0.09775090217590332,\"isRecurrent\":false},{\"fromNeuron\":2,\"toNeuron\":7,\"weight\":1.0683479309082031,\"isRecurrent\":false},{\"fromNeuron\":2,\"toNeuron\":9,\"weight\":2.705871105194092,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":5,\"type\":0,\"neuronID\":2,\"activationResponse\":1,\"splitX\":0.5714285969734192,\"splitY\":0},{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":3,\"toNeuron\":6,\"weight\":-0.4662399888038635,\"isRecurrent\":false},{\"fromNeuron\":3,\"toNeuron\":7,\"weight\":3.3073086738586426,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":1.3970988988876343,\"type\":0,\"neuronID\":3,\"activationResponse\":0.9197852611541748,\"splitX\":0.7142857313156128,\"splitY\":0},{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":4,\"toNeuron\":6,\"weight\":-0.8872920274734497,\"isRecurrent\":false},{\"fromNeuron\":4,\"toNeuron\":7,\"weight\":-0.9433706402778625,\"isRecurrent\":false},{\"fromNeuron\":4,\"toNeuron\":9,\"weight\":0.8649682402610779,\"isRecurrent\":false},{\"fromNeuron\":4,\"toNeuron\":39,\"weight\":-1.6333357095718384,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":0.6981958746910095,\"type\":0,\"neuronID\":4,\"activationResponse\":1,\"splitX\":0.8571429252624512,\"splitY\":0},{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":5,\"toNeuron\":6,\"weight\":0.8984636068344116,\"isRecurrent\":false},{\"fromNeuron\":5,\"toNeuron\":7,\"weight\":-0.8724162578582764,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":1,\"type\":2,\"neuronID\":5,\"activationResponse\":1,\"splitX\":0.1428571492433548,\"splitY\":0},{\"linksTo\":[{\"fromNeuron\":0,\"toNeuron\":6,\"weight\":0.3998056650161743,\"isRecurrent\":false},{\"fromNeuron\":1,\"toNeuron\":6,\"weight\":1.3139477968215942,\"isRecurrent\":false},{\"fromNeuron\":2,\"toNeuron\":6,\"weight\":0.09775090217590332,\"isRecurrent\":false},{\"fromNeuron\":3,\"toNeuron\":6,\"weight\":-0.4662399888038635,\"isRecurrent\":false},{\"fromNeuron\":4,\"toNeuron\":6,\"weight\":-0.8872920274734497,\"isRecurrent\":false},{\"fromNeuron\":5,\"toNeuron\":6,\"weight\":0.8984636068344116,\"isRecurrent\":false},{\"fromNeuron\":7,\"toNeuron\":6,\"weight\":-0.5298008918762207,\"isRecurrent\":false},{\"fromNeuron\":9,\"toNeuron\":6,\"weight\":-0.15490210056304932,\"isRecurrent\":false}],\"linksFrom\":[{\"fromNeuron\":6,\"toNeuron\":7,\"weight\":-1.1464767456054688,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":0.7721230983734131,\"type\":3,\"neuronID\":6,\"activationResponse\":1,\"splitX\":0.3333333432674408,\"splitY\":1},{\"linksTo\":[{\"fromNeuron\":1,\"toNeuron\":7,\"weight\":-1.841086983680725,\"isRecurrent\":false},{\"fromNeuron\":2,\"toNeuron\":7,\"weight\":1.0683479309082031,\"isRecurrent\":false},{\"fromNeuron\":3,\"toNeuron\":7,\"weight\":3.3073086738586426,\"isRecurrent\":false},{\"fromNeuron\":4,\"toNeuron\":7,\"weight\":-0.9433706402778625,\"isRecurrent\":false},{\"fromNeuron\":5,\"toNeuron\":7,\"weight\":-0.8724162578582764,\"isRecurrent\":false},{\"fromNeuron\":9,\"toNeuron\":7,\"weight\":-0.3332200050354004,\"isRecurrent\":false},{\"fromNeuron\":6,\"toNeuron\":7,\"weight\":-1.1464767456054688,\"isRecurrent\":false},{\"fromNeuron\":39,\"toNeuron\":7,\"weight\":0.11120444536209106,\"isRecurrent\":false}],\"linksFrom\":[{\"fromNeuron\":7,\"toNeuron\":6,\"weight\":-0.5298008918762207,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":0.9941025376319885,\"type\":3,\"neuronID\":7,\"activationResponse\":1,\"splitX\":0.6666666865348816,\"splitY\":1},{\"linksTo\":[{\"fromNeuron\":0,\"toNeuron\":9,\"weight\":0.5525347590446472,\"isRecurrent\":false},{\"fromNeuron\":4,\"toNeuron\":9,\"weight\":0.8649682402610779,\"isRecurrent\":false},{\"fromNeuron\":2,\"toNeuron\":9,\"weight\":2.705871105194092,\"isRecurrent\":false},{\"fromNeuron\":1,\"toNeuron\":9,\"weight\":0.33449316024780273,\"isRecurrent\":false},{\"fromNeuron\":39,\"toNeuron\":9,\"weight\":0.16371417045593262,\"isRecurrent\":false}],\"linksFrom\":[{\"fromNeuron\":9,\"toNeuron\":7,\"weight\":-0.3332200050354004,\"isRecurrent\":false},{\"fromNeuron\":9,\"toNeuron\":6,\"weight\":-0.15490210056304932,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":0.9999996423721313,\"type\":1,\"neuronID\":9,\"activationResponse\":1,\"splitX\":0.2857142984867096,\"splitY\":0},{\"linksTo\":[{\"fromNeuron\":4,\"toNeuron\":39,\"weight\":-1.6333357095718384,\"isRecurrent\":false},{\"fromNeuron\":0,\"toNeuron\":39,\"weight\":-1.4688773155212402,\"isRecurrent\":false}],\"linksFrom\":[{\"fromNeuron\":39,\"toNeuron\":9,\"weight\":0.16371417045593262,\"isRecurrent\":false},{\"fromNeuron\":39,\"toNeuron\":7,\"weight\":0.11120444536209106,\"isRecurrent\":false}],\"sumActivation\":0,\"output\":0.09952177852392197,\"type\":1,\"neuronID\":39,\"activationResponse\":1,\"splitX\":0.8571429252624512,\"splitY\":0}],\"depth\":2}";

JsonObject& root = jsonBuffer.parseObject(json); // can I do this at compile time?

if (!root.success()) {
    Serial.println("parseObject() failed");
    return;
  }

JsonArray& neurons = root["neurons"];

size_t sizeNeurons = neurons.size();
Serial.println(sizeNeurons);

JsonObject& neurons0 = neurons[0];

JsonArray& neurons0_linksFrom = neurons0["linksFrom"];

size_t size_linksFrom = neurons0_linksFrom.size();
Serial.println(size_linksFrom);

JsonObject& neurons0_linksFrom0 = neurons0_linksFrom[0];
int neurons0_linksFrom0_fromNeuron = neurons0_linksFrom0["fromNeuron"]; // 0
int neurons0_linksFrom0_toNeuron = neurons0_linksFrom0["toNeuron"]; // 6
float neurons0_linksFrom0_weight = neurons0_linksFrom0["weight"]; // 0.3998056650161743
bool neurons0_linksFrom0_isRecurrent = neurons0_linksFrom0["isRecurrent"]; // false

JsonObject& neurons0_linksFrom1 = neurons0_linksFrom[1];
int neurons0_linksFrom1_fromNeuron = neurons0_linksFrom1["fromNeuron"]; // 0
int neurons0_linksFrom1_toNeuron = neurons0_linksFrom1["toNeuron"]; // 9
float neurons0_linksFrom1_weight = neurons0_linksFrom1["weight"]; // 0.5525347590446472
[... cut ...]
int neurons3_splitY = neurons3["splitY"]; // 0

int depth = root["depth"]; // 2

Serial.println(depth);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
**Compiler output:**
```
Sketch uses 12266 bytes (38%) of program storage space. Maximum is 32256 bytes.

Global variables use 5678 bytes (277%) of dynamic memory, leaving -3630 bytes for local variables. Maximum is 2048 bytes.
```

Anyway to use the 32KB of program storage space to store the deserialized JSON at compile time?

bblanchon, 2018-01-02:
Hi @Linkaan,

Indeed ArduinoJson can parse a JSON document stored in Flash.
You just need to move the string in Flash memory and make sure it has the type `__FlashStringHelper*`.
The simplest way is:

```c++
auto json = F("{\"neurons\":[{\"linksTo\":[],\"linksFrom\":[{\"fromNeuron\":0,\"toNeuron\"...
```

However, since the input in read-only, the parse will copy the strings in memory.
The [assistant](https://arduinojson.org/assistant/) says that this duplication requires 4522 bytes, which is still too big for the UNO.
(btw, the chapter [Deserialize with ArduinoJson](https://arduinojson.org/book/) covers all of that)

If I were you, I would only store the binary matrix on the microcontroller.
The secret to a successful embedded project is to perform the maximum amount of work at *compile-time*, to reduce the work at *run-time*.
You can write a small program to generate the binary matrix from the JSON document on your computer, and you can even use ArduinoJson to do that.

Regards,
Benoit

Linkaan, 2018-01-03:
Thanks for the quick answer, I understand that the strings will have to be stored in memory. I ended up using a raspberry pi for running the neural network and using ArduinoJson to communicate the inputs and outputs. I'll keep the binary matrix idea in mind though for upcoming projects :)

closed 2018-01-03 by Linkaan