bblanchon/ArduinoJson #656: I can´t decode an array

Matelog, 2018-01-02:
Hi everyone. I have an issue in the success of parsing, I´ve increased the buffer and nothing happened.  Below the code. 

`#include <ESP8266WiFi.h>        // ESP8266 WIFI Lib
#include <ESP8266WiFiMulti.h>   // WifiMulti Lib for connection handling
#include <ESP8266HTTPClient.h>  // HTTPClient for web requests
#include <ArduinoJson.h>        // JSON Parser

WiFiClient UsuarioWifi;
ESP8266WiFiMulti WiFiMulti;

char* WIFI_SSID = "Speedy-CC2F12";
char* WIFI_PASS = "4578823057";

String OWM_API_KEY = "f44ce8aa2623eeb307f7d5fb56faa6a3";// Open Weather Map API Key
String OWM_CITY_ID = "3699223";           // Open Weather Map CityID
String OWM_FIELD   = "temp";       // can be "temp","pressure","humidity","temp_min", or "temp_max"
String OWM_UNITS   = "metric";   // can be "imperial", "metric", or "kelvin"

void setup() {
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
}

void loop() {
  checkOWM();
  delay(30000);
}
void checkOWM() {
  StaticJsonBuffer<2000> jsonBuffer;      // JSON Parser setup
  // wait for WiFi connection
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    Serial.println("Conexion con éxito");
    HTTPClient http;
    http.begin("http://api.openweathermap.org/data/2.5/forecast/daily?q=Lucca&mode=json&&APPID=f44ce8aa2623eeb307f7d5fb56faa6a3");
    int httpCode = http.GET();
    if (httpCode > 0) {
      if (httpCode == HTTP_CODE_OK) {
        String payload = http.getString();
        Serial.println("Esto es el payload/ http.getString() ");
        Serial.println(payload);
        JsonObject& owm_data = jsonBuffer.parseObject(payload);

        int field = owm_data[0]["city"];
        int code = owm_data["weather"][0]["id"];
        
        Serial.println("El id de Lucca es: ");
        Serial.println(field);
        
        if (!owm_data.success()) {
          Serial.println("Parsing failed");
          return;
        }   
      }
    }
    http.end();
  }
}

`

bblanchon, 2018-01-03:
Hi @Matelog,

For the compilation error, please see:
[Error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'](https://arduinojson.org/faq/error-equals-is-not-a-member-of/)

For the parsing error, try to use a [DynamicJsonBuffer](https://arduinojson.org/faq/what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer/) and use the [assistant](https://arduinojson.org/assistant/) to compute its size.

BTW, there is a project using OpenWeatherMap in the "Case Studies" chapter of [Mastering ArduinoJson](https://arduinojson.org/book/).

Regards,
Benoit

Matelog, 2018-01-07:
Hi @bblanchon !
```
#include <ESP8266WiFi.h>        // ESP8266 WIFI Lib
#include <ESP8266WiFiMulti.h>   // WifiMulti Lib for connection handling
#include <ESP8266HTTPClient.h>  // HTTPClient for web requests
#include <ArduinoJson.h>        // JSON Parser

WiFiClient UsuarioWifi;
ESP8266WiFiMulti WiFiMulti;

char* WIFI_SSID = "Speedy-CC2F12";
char* WIFI_PASS = "457";
void setup() {
  Serial.begin(115200);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
}

void loop() {
  TempMax();
  delay(30000);
}
void TempMax() {
  DynamicJsonBuffer jsonBuffer (4000);      // JSON Parser setup
  // wait for WiFi connection
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
  if ((WiFiMulti.run() == WL_CONNECTED)) {
    Serial.println("Conexion con éxito");
    HTTPClient http;
    
    http.begin("http://api.openweathermap.org/data/2.5/forecast/daily?q=Buenos%20Aires&mode=json&&APPID=f44ce8aa2623eeb");
    int httpCode = http.GET();
    if (httpCode > 0) {
      if (httpCode == HTTP_CODE_OK) {
        String payload = http.getString();
        Serial.println("Esto es el payload/ http.getString() ");
        Serial.println(payload);
        JsonObject& root = jsonBuffer.parseObject(payload);
        
        JsonArray& list = root["list"];
        int temp_max_week[7];
        float alt = 0;
        JsonObject& lista0 = list[0];//All the week here.
        JsonObject& lista1 = list[1];
        JsonObject& lista2 = list[2];
        JsonObject& lista3 = list[3];
        JsonObject& lista4 = list[4];
        JsonObject& lista5 = list[5];
        JsonObject& lista6 = list[6];
```
**I tried to do that with for but I couldn´t**
```
        JsonObject& lista_temp0 = lista0["temp"];
        JsonObject& lista_temp1 = lista1["temp"];
        JsonObject& lista_temp2 = lista2["temp"];
        JsonObject& lista_temp3 = lista3["temp"];
        JsonObject& lista_temp4 = lista4["temp"];
        JsonObject& lista_temp5 = lista5["temp"];
        JsonObject& lista_temp6 = lista6["temp"];

        temp_max_week[0] = lista_temp0["max"];
        temp_max_week[1] = lista_temp1["max"];
        temp_max_week[2] = lista_temp2["max"];
        temp_max_week[3] = lista_temp3["max"];
        temp_max_week[4] = lista_temp4["max"];
        temp_max_week[5] = lista_temp5["max"];
        temp_max_week[6] = lista_temp6["max"];
        for(int j = 0; j < 7; j++){
[... 3 lines cut ...]
        }
        Serial.println("The max temperature in the week:");
        Serial.println(alt);
        if (!root.success()) {
          Serial.println("Parsing failed");
          return;
        }   
      }
    }
    http.end();
  }
}

```
We´ve already done it. There is a problem in the json, I think. When we get the array in arduino we have in the max temperature of index 0 in list : 294
but in the real array : 21.62

## Before of arduino:
`{"city":{"id":3435910,"name":"Buenos Aires","coord":{"lon":-58.4371,"lat":-34.6076},"country":"AR","population":1000000},"cod":"200","message":0.0487987,"cnt":7,"list":[{"dt":1515254400,"temp":{"day":21.62,"min":20.77,"max":21.62,"night":20.77,"eve":21.62,"morn":21.62},"pressure":1029.11,"humidity":74,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01n"}],"speed":3.91,"deg":45,"clouds":0},{"dt":1515340800,"temp":{"day":26.41,"min":20.14,"max":28.55,"night":23.67,"eve":28.55,"morn":20.14},"pressure":1030.12,"humidity":69,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":7.16,"deg":337,"clouds":0},{"dt":1515427200,"temp":{"day":27.84,"min":22.22,"max":29.25,"night":23.52,"eve":29.01,"morn":22.22},"pressure":1032.13,"humidity":65,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":5.66,"deg":355,"clouds":0},{"dt":1515513600,"temp":{"day":32.01,"min":25.53,"max":32.01,"night":25.53,"eve":28.41,"morn":26.63},"pressure":1028.78,"humidity":0,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":5.28,"deg":77,"clouds":0},{"dt":1515600000,"temp":{"day":31.59,"min":25.98,"max":31.59,"night":25.98,"eve":28.53,"morn":26.71},"pressure":1026.57,"humidity":0,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":5.61,"deg":89,"clouds":17},{"dt":1515686400,"temp":{"day":32.92,"min":25.85,"max":32.92,"night":25.85,"eve":28.68,"morn":28.86},"pressure":1023.13,"humidity":0,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":4.11,"deg":74,"clouds":2},{"dt":1515772800,"temp":{"day":32.79,"min":24.07,"max":32.79,"night":24.07,"eve":28.61,"morn":27.14},"pressure":1019.77,"humidity":0,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":4.07,"deg":78,"clouds":1}]}`

## After Arduino:
`{"city":{"id":3435910,"name":"Buenos Aires","coord":{"lon":-58.4371,"lat":-34.6076},"country":"AR","population":1000000},"cod":"200","message":10.982991,"cnt":7,"list":[{"dt":1515254400,"temp":{"day":294.8,"min":294.8,"max":294.8,"night":294.8,"eve":294.8,"morn":294.8},"pressure":1029.42,"humidity":78,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01n"}],"speed":5.12,"deg":36,"clouds":0},{"dt":1515340800,"temp":{"day":299.82,"min":293.86,"max":301.7,"night":296.82,"eve":301.7,"morn":293.86},"pressure":1030.12,"humidity":69,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":7.16,"deg":337,"clouds":0},{"dt":1515427200,"temp":{"day":300.99,"min":29 [... cut ...] in":299,"max":306.07,"night":299,"eve":301.83,"morn":302.01},"pressure":1023.13,"humidity":0,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":4.11,"deg":74,"clouds":2},{"dt":1515772800,"temp":{"day":305.94,"min":297.22,"max":305.94,"night":297.22,"eve":301.76,"morn":300.29},"pressure":1019.77,"humidity":0,"weather":[{"id":800,"main":"Clear","description":"sky is clear","icon":"01d"}],"speed":4.07,"deg":78,"clouds":1}]}`

Thanks for help !! 
Regards, Lautaro

bblanchon, 2018-01-10:
Hi Lautaro,

I'm sorry, I've read your message about ten times, but I don't understand.
Can you explain the problem differently?

Regards,
Benoit

Matelog, 2018-01-11:
Hello @bblanchon !
When the array get into arduino is: `"morn":21.62`
but when get out is:`"morn":294.8`
Is it a mistake of the code? It happen because I requested the array in celsius but when we process the array we can see in the serial monitor the array in kelvin grades.
We can transform degrees from kevin to celsius but I think it would be more comfortable not to do it. 
Sorry for my english,
Thanks Benoit
Regards !

bblanchon, 2018-01-11:
Hi Lautaro,

The [OpenWeatherMap documentation](https://openweathermap.org/current#data) says you need to add `&units=metric` to get the temperature in Celcius.

I'm closing this issue as it's not related to ArduinoJson; thank you for your understanding.

Regards,
Benoit

closed 2018-01-11 by bblanchon

Matelog, 2018-01-13:
Ok.Thanks!!