bblanchon/ArduinoJson #657: Unable to compile program using make file

dharshanv07, 2018-01-04:
I'm trying to compile my program for nodeMCU esp8266 using a make file but it is throwing error and the compilation is getting terminated, though the program can be successfully compiled from arduino ide 1.8.5. Using libraries like aJson the program is getting compiled successfully from both the make file and arduino ide.
Error:
parseInteger.cpp
/root/Arduino/libraries/ArduinoJson/test/IntegrationTests/gbathree.cpp:6:21: fatal error: catch.hpp: No such file or directory
 #include <catch.hpp>
                     ^
parseFloat.cpp
compilation terminated.
/root/Arduino/libraries/ArduinoJson/test/IntegrationTests/round_trip.cpp:6:21: fatal error: catch.hpp: No such file or directory
 #include <catch.hpp>
                     ^
compilation terminated.

bblanchon, 2018-01-04:
Hi @dharshanv07,

You're currently compiling the test suite, which is probably not what you want.

ArduinoJson is a [header-only library](https://en.wikipedia.org/wiki/Header-only), it doesn't need to be compiled.
You can even [download ArduinoJson as a single file](https://github.com/bblanchon/ArduinoJson/releases/download/v5.12.0/ArduinoJson-v5.12.0.h).

See also:

* [User Manual: Installation](https://arduinojson.org/doc/installation/)

Regards,
Benoit

dharshanv07, 2018-01-04:
Thanks @bblanchon for your quick reply.But could u please tell me how to solve this problem. if possible Step by step as i could not understand what you told.

bblanchon, 2018-01-05:
Hi @dharshanv07,

Again, you should not compile ArduinoJson because it is a header-only library.
You need to remove any reference to ArduinoJson from the Makefile.

Regards,
Benoit

dmcinnes, 2018-01-06:
I've had the same problem, using [makeEspArduino](https://github.com/plerup/makeEspArduino), it seems to want to compile everything.
I removed the `test`, `fuzzing` and `third-party` directories in the ArduinoJson library installed through the Arudino app (found at ~/Documents/Arduino/libraries/ArduinoJson on my Mac) and I was able to get it to work.
There's probably a more elegant way of telling it to only use the main header, but at least I'm unblocked now ;)

bblanchon, 2018-01-07:
I never use `makeEspArduino`, but apparently, the proper way to deal with this is to set the variable `EXCLUDE_DIRS `.
See https://github.com/plerup/makeEspArduino/issues/47#issue-245733864

closed 2018-01-14 by bblanchon