bblanchon/ArduinoJson #658: Force char[] or const char * to be duplicated to JsonObject

ttlappalainen, 2018-01-06:
JsonObject uses as default reference to values set by char[]. Is there way to force value to be duplicated? If I e.g. use JsonObject as reference getting class description:
class MyClass : public BaseClass {
...
  virtual void GetJson(JsonObject &Json);
};

and inside GetJson I have to generate some results:
void GetJson(JsonObject &Json) {
  char Generated[100];
  // do something to set content of Generated
  Json[strGeneratedKey]=Generated;
};

I could not find good way to force value to be copied to buffer instead of referencing it, which does not work here. I can go around with Json[strGeneratedKey]=String(Generated);, which forces copy, but also makes extra work with String constructing etc. Also I do not want to carry JsonBuffer through a parameter list so that I could use strdup.

So is there some other way, which I did not see. And if not, would it be possible to add? Or does my basic idea to use JsonObject as reference paramter have some misunderstanding?

bblanchon, 2018-01-07:
Hi @ttlappalainen,

It's funny you ask about this feature now, because it was my secret plan for v5.13.

The syntax would look like the following:

```c++
char tmp[256];
snprintf(tmp, 256, "File %d", fileNum);
root["desc"] = VolatileString(tmp);
```

So, you won't need to call `strdup()` explicitly; instead, you would just have to wrap the string with the type `VolativeString`.

What do you think?

Regards,
Benoit

ttlappalainen, 2018-01-07:
I just started to use your great library 2 months ago. I noticed the behaviour earlier, but could go around it with other arrangement, so I did not thought it so important at the beginning.

That would be OK - something like that I was looking for. On the orher hand VolatileString does not look very descriptive for me, since it does not show that function has something to do with your library. One could think that it is just an other string class. Something like ArduinoJson::Dup(tmp), would tell more.

bblanchon, 2018-01-08:
Everything in the library is declared in the ArduinoJson namespace, but `ArduinoJson.h` makes a `using namespace`. If you want to avoid that, use `ArduinoJson.hpp`.

`Dup()` isn't a good name either because there is not always duplication.
For example:

```c++
root[Dup(tmp)] = 42;
```

If the key is already present in the object, the string is not duplicated.

That's why I chose `VolatileString()`, but I'm still open to suggestions.

bblanchon, 2018-01-08:
I wonder I we take this problem at the right angle.

What if we say that `char[]` and `const char*` are duplicated by default?
We could wrap them with a `NoDup` type to prevent duplication.
The only difficulty is to detect the string literals ([which seems to be possible](https://wandbox.org/permlink/fAsDRpr70bNGJFwE)) because we don't want to duplicate them.

What do you think?

ttlappalainen, 2018-01-08:
Would be easy, but how does it effect to down compatibility and existing code? Suddently all chars will be duplicated, so if someone has buffer just near the limit (or long const char*:s), it will crash ones program. One solution would be to have compiler definition or setting e.g. JsonBuffer.DupCharAsDefault=true to change default behaviour.

VolatileString() is good hit. I did not check that it has using ArduinoJson. Also I did not thought the key side.

bblanchon, 2018-01-11:
I think we can limit the impact of the change by duplicating only `char[]` by default, which would work for your case.

Then, we could have two magic functions `NoDup()` and `ForceDup()`.
Since the default behavior is the right one now, they should be rarely used.

I don't want to have a configurable default as it will require testing everything twice.

ttlappalainen, 2018-01-11:
Sounds reasonable solution. 

How does that work with keys? If the key is type char [] and same variable will be used several times (e.g. with array), does it automatically make a dup for the first and rest points to the first?

Is there way force dup for all strings:
Json.DupKeys(); Json.DupStrings();?

I am just thinking forward and not sure is this necessary. But that could be necessary, if I use as in my original question:
void GetJson(JsonObject &Json) {
  Json["Title"]=GetTitle(); // GetTitle(); returns const char, but I do not want to dup it yet to save memory
};
...
... {
  MyObject.GetJson(Json);
  ...
  // Here I find out that object is not needed, but I still need Json
  Json.DupStrings();
  delete MyObject;

In PC systems I would probably automatically dup all, but with these small memory systems it may be important to save memory and dup later if is necessary.

bblanchon, 2018-01-12:
I came up with a new idea that fits every need:

| type | duplication |
|:----|:---|
| char* |  yes |
| const char* | no |
| char[] |  yes |
| const char[] | no |
| String | yes |
| std::string | yes |

With this scheme, we don't even need `Dup()` and `NoDup()`, because casting is sufficient.
For example, you have a `char[]` value that you don't want to duplicate, just cast it to `const char*`:

```c++
char tmp[256];
snprintf(tmp, 256, "File %d", fileNum);
root["desc"] = (const char*)tmp; // cast to prevent duplication
```

The default behavior is the right one for most cases, and I can finally get rid of [this page](https://arduinojson.org/faq/using-a-loop-why-are-values-all-the-same/).

ttlappalainen, 2018-01-12:
I think you still need Dup. Think about case:
root["Name"]=MyClass.GetName();  // GetName returns const char *
delete MyClass;  // Now I am in trouble with root
In above casting:
root["Name"]=(char *)MyClass.GetName();
is just not good programming.

On other points this looks simple solution.

bblanchon, 2018-01-14:
On the one hand, `root["key"]=Dup(tmp)` is more expressive than `root["key"]=(char*)tmp`.
But on the other hand, it requires to learn new concepts for `Dup()` and `DoDup()`.

I think it's better to stick to casting because it's just simpler to understand.

Anyway, now that the default behavior is the one people expect, people will rarely see the problem.

(btw, calling `delete` is not good programming either 😛 )

commit e92612b5 (2018-01-14): Changed the rules of string duplication (fixes #658)
closed 2018-01-14 by bblanchon