bblanchon/ArduinoJson #659: ArduinoJson not working outside of main file?

ResaloliPT, 2018-01-07:
Hello im getting something weird i dont know if its IDE related or Library related (Can be my code but not telling me tho).

I have installed Arduino IDE 1.8.5 (Lastest as of 06/01/18)

Bellow i have 2 zips that can be downloaded and run with Arduino IDE just for easy
[CocktailMachine (With Errors).zip](https://github.com/bblanchon/ArduinoJson/files/1609403/CocktailMachine.With.Errors.zip)
[CocktailMachine (No Errors).zip](https://github.com/bblanchon/ArduinoJson/files/1609404/CocktailMachine.No.Errors.zip)

When my code (See bellow) is in main file with the setup() and loop() the Arduino IDE compile sucefully but when i cut and paste onto a diferent file in same project folder aside of main.ino it start trowing out errors.

Those are the compile errors:

Handlers:75: error: 'JsonObject' does not name a type

 JsonObject& readCard() {

 ^

Handlers:82: error: variable or field 'saveCard' declared void

 void saveCard(JsonObject& buff){

               ^

Handlers:82: error: 'JsonObject' was not declared in this scope

Handlers:82: error: 'buff' was not declared in this scope

 void saveCard(JsonObject& buff){

                           ^

exit status 1
'JsonObject' does not name a type

And here is the code in question:

```/// Bibliotecas
#include <ArduinoJson.h>
#include <SD.h>

/// Definiçoes
DynamicJsonBuffer jsonBuffer; // Criar armazenamento de JSON

/// Variaveis
int hasChanged = 1; // Se foi alterado (Default: 1  para fazer sempre uma primeira leitura)
String filename = "data.txt";
File dataFile;
/*
Estrutura JSON
{
  "Cocktails": { // Guarda os  cocktails
    1: {
      "dID": 1,
      "dName": "Cocktail 1",
      "dMix": [[0,3],[1,5],[2,6],[3,2]]
    },
    2: {
      "dID": 2,
      "dName": "Cocktail 1",
      "dMix": [[0,3],[1,5],[2,6],[3,2]
    }
  },
  "Settings": { // Guarda possiveils defenições
    
  },
  "Storage": { // Guarda Quantidades das bebidas nas garrafas
    "1": {
      "bID: 1,
      "bName": "Jack&Daniels",
      "bCL": 100,
      "bCLRemain": 36
    }
  }
}
*/

/// Structs
struct Settings {
  String setting;
  String value;
};

struct Storage {
  int bID;
  String bName;
  float bCL;
  float bCLRemain;
};

struct Mix {
  int bID;
  int bPer;
};

struct Cocktails{
  int dID;
  String dName;
  int dMix[4];
};

struct Config {
  Cocktails cocktails[];
  Settings settings[];
  Storage storage[];
};

/// Functions

//Utils
JsonObject& readCard() {
  dataFile = SD.open(filename);
  JsonObject& data = jsonBuffer.parseObject(dataFile);
  dataFile.close();
  return data;
}

void saveCard(JsonObject& buff){
  SD.remove(filename);
  dataFile = SD.open(filename, FILE_WRITE);
  buff.printTo(dataFile);
  dataFile.close();
}

//Storage
void getStorage(struct Storage *bottlesList){
  JsonObject& root = readCard();
  JsonObject& bottles = root["Storage"];
  int nbottles = bottles.size();
    
  for(int i = 0;i < nbottles;i++){
    JsonObject& selBottle = bottles[(String)i];
    bottlesList[i].bCLRemain = selBottle.get<float>("bCLRemain");
    bottlesList[i].bCL = selBottle.get<float>("bCL");
    bottlesList[i].bName = selBottle.get<float>("bName");
    bottlesList[i].bID = selBottle.get<int>("bID");
  }
}

void reduceStorage(int bottleID, float usedCL){
  JsonObject& root = readCard();
  JsonObject& bottles = root["Storage"];
  JsonObject& selBottle = bottles[(String)bottleID];
  float remainingCL = selBottle.get<float>("bCLRemain");
  remainingCL -= roundf(usedCL);
  selBottle.set("bCLRemain", remainingCL);
  saveCard(root);
}

void refillStorage(int bottleID){
  JsonObject& root = readCard();
  JsonObject& bottles = root["Storage"];
  JsonObject& selBottle = bottles[(String)bottleID];
  
  selBottle.set("bCLRemain", selBottle["bCL"]);
  saveCard(root);
}

//Cocktails
struct Cocktails getCocktail(int cocktailID){
  JsonObject& root = readCard();
  JsonObject& cocktails = root["Cocktails"];
  JsonObject& selCocktail = cocktails[(String)cocktailID];
  JsonArray& selMix = selCocktail["mix"];
  
  struct Cocktails cocktail;
  selMix.copyTo(cocktail.dMix);
  cocktail.dName = selCocktail.get<String>("name");
  cocktail.dID = cocktailID;
  
  return cocktail;
}

void getDrinks(struct Cocktails *CocktailList){
  JsonObject& root = readCard();
  JsonObject& cocktails = root["Cocktails"];
  int nCocktails = cocktails.size();
  
  for(int i = 0;i < nCocktails;i++){
    JsonObject& selCocktail = cocktails[(String)i];
    JsonArray& selMix = selCocktail["dMix"];
    selMix.copyTo(CocktailList[i].dMix);
    CocktailList[i].dName = selCocktail.get<String>("dName");
    CocktailList[i].dID = selCocktail.get<int>("dID");
  }
}

void addDrink(){
  
}

void remDrink(int drink){
  
}

void updateDrink(){
  
}

//Settings
void setSetting(String setting, String value){
  
}

void getSetting(){
  
}

bblanchon, 2018-01-08:
Hi @ResaloliPT,

This behavior is a curiosity of the Arduino IDE: it scans the main file to detect the libraries to include.
So you need to `#include` all libraries in the main file even if you don't use them there.

For example, to fix your sketch, just add the following lines at the top of `CocktailMachine.ino`:

```c++
#include <ArduinoJson.h>
#include <SD.h>
```

Regards,
Benoit

ResaloliPT, 2018-01-08:
Compiling as expected, looks like i did a beginner in arduino error --' 

*Note to self: all arduino includes in main sketch*

Thx for the help!

closed 2018-01-08 by ResaloliPT

bblanchon, 2018-01-09:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)

dtekluva, 2018-09-20:
#include <ArduinoJson.h>
please why do you think i am getting root does not name a type error in my code,

  const size_t bufferSize = JSON_OBJECT_SIZE(5);
  DynamicJsonBuffer jsonBuffer(bufferSize);
  
  JsonObject& root = jsonBuffer.createObject();
  root["co"] = 196;
  root["ch4"] = 125;
  root["aq"] = 211;
  root["h"] = 95;
  root["time"] = time_stamp;
  
  root.printTo(Serial);

i put the code as is given on the website

bblanchon, 2018-09-21:
@dtekluva, the code looks good; maybe you install Arduino version 6 instead of 5.