bblanchon/ArduinoJson #661: How can i receive special characters like +,-,%,& in json?

parmarravi, 2018-01-09:
I am trying to send  this json message .However i am not able to get any special characters, instead i am getting blank space.
> {
>   "Message":{
>   "Mes":"hello +world 50%"
>}
>   }
how can i get special characters ?

bblanchon, 2018-01-10:
Hi @parmarravi,

I cannot reproduce the problem.
See demo here: https://wandbox.org/permlink/HQLtkD6QOD3WWdCj

Can you provide an [MCVE](https://stackoverflow.com/help/mcve)?

Regards,
Benoit

parmarravi, 2018-01-10:
Hi @**bblanchon** ,
Thank you for your reply
This is my code.
```
/*

json data structure 

{ 
  "data":"hello+world get 50%"
}

*/
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ArduinoJson.h>

String Jsondata;

const char* ssid = "........";
const char* password = "........";

ESP8266WebServer server(80);
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println (WiFi.localIP());
  server.on("/", handleBody);
  server.begin();
  Serial.println("Server started");
}

void loop() {
  // put your main code here, to run repeatedly:
  server.handleClient();
}
void handleBody() { //Handler for the body path
  if (server.hasArg("plain") == false) { //Check if body received
    server.send(404, "text/plain", "error");
    return;
  }
  DynamicJsonBuffer  jsonBuffer;
  String jsondata = server.arg("plain");
  Serial.println(jsondata);
  JsonObject& root = jsonBuffer.parseObject(jsondata);
  if (!root.success()) {
    server.send(404, "text/plain", "error");
  } else {
    server.send(200, "text/plain", "ok");
  }
}
```

parmarravi, 2018-01-10:
Hi, @bblanchon ,
I used  url encoding and the problem got solved.
For example : "%" with "%25" and  "+" with %2B .
This is the site i referred `https://www.w3schools.com/tags/ref_urlencode.asp`

closed 2018-01-10 by parmarravi