bblanchon/ArduinoJson #664: How to add nested object in iteration?

Modifikator, 2018-01-12:
```

Input obj:

String _jsonSensors = {"sensors":[{"sensor":"ds18b20","pin":"4"},...]}

Example:

  DynamicJsonBuffer jsonBuffer;
  JsonObject& oSENSORS = jsonBuffer.parseObject(_jsonSensors);
  JsonArray& aSensors = oSENSORS["sensors"].asArray();
  //...
  
 String sensor;
  
for (auto iSensor: aSensors) {

	sensor = iSensor["sensor"].as<const char*>();
   
   //... 
   
     if(sensor == "ds18b20") {
	   
	   //  attempt one:
	   
	   StaticJsonBuffer<200> jsonBufferDs;
	   JsonObject& value = jsonBufferDs.createObject();
	   
	   for(auto dsTemp: dsThermometers) {
		//...
		value[address] = temp;
	   }
	   iSensor.set("values",value); 
	   
	   // attempt two:
	   
	   JsonObject& value = iSensor.createNestedObject("values");
	   
	   for(auto dsTemp: dsThermometers) {
		//...	
		value[address] = temp;
	   }
   }
}

I need output:

{"sensors":[{"sensor":"ds18b20","pin":"4","values":{"abcd":"22.5","efgk":"-2.0"}}]}
```

Modifikator, 2018-01-12:
So it works, but how true is it, can there be a more correct code?
```
for (int i = 0; i <= sz - 1; i++) { 
sensor = aSensors[i]["sensor"].as<const char*>();
 if(sensor == "ds18b20") {      
      StaticJsonBuffer<200> jsonBufferDs;
     for(auto dsTemp: dsThermometers) {
          value += "\"" + address + "\":\"" + (String)temp + "\",";
     }
    aSensors[i]["values"] = jsonBufferDs.parseObject("{" + value.substring(0, value.lastIndexOf(",")) + "}");
 }
}
```

Modifikator, 2018-01-12:
Very good, thanks :)

closed 2018-01-12 by Modifikator

bblanchon, 2018-01-12:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)