bblanchon/ArduinoJson #666: Multiple json buffer used

IoTAll, 2018-01-13:
Hi!
Thanks a LOT for this great library, having a lot of fun with it with different projects!
I feel really stuck with an issue right now, would love to have some help to finalize my project.

I am using an ESP32 and I have been successfully downloading and parsing weather data using your ArduinoJsonlibrary from Wunderground, even the big hourly forecast. (73884 bytes simulated on your ArduinoJson Assistant for ESP8266). This is super nice!!

In my sketch I created separate functions for each http request, json buffer creation, parsing, for each request I do: one function for conditions, one for forecast and one for hourly forecast. 

Where I am stuck is: I tried now to add another function, using the same way as the other ones, requesting my geolocation through the GoogleMap API (I don't know if this matters but this one is using a WifiSecure connection)

This works great too but...
If I run the functions in this order: 
1) GoogleMAP API function
2) conditions function
3) forecast function
4) hourly function
everything run fine!.... aside the hourly forecast function. I get the "parsing fail" message.

I first thought it was a RAM memory problem but then doing some tests, I saw that if I put the hourly forecast function first, and then the GoogleMap API Geolocation function after, all works good again!

but what I would like is to geolocate my device first, before requesting and parsing the weather information.

I was wondering if I maybe have a json buffer creation or size issue?
In each function, I create the json buffer like this:
```
  const size_t capacity = XXXXX;
  DynamicJsonBuffer jsonBuffer(capacity);

  // Parse JSON object
  JsonObject& root = `jsonBuffer.parseObject(client);`
```
Here are approximatly the different json buffer sizes I create are in each function: (calculated on ArduinoJson assistant, for ESP8266)
geolocation function: DynamicJsonBuffer size: 139
conditions function: DynamicJsonBuffer size: 3389
forecast function: DynamicJsonBuffer size: 11173
hourly  function: DynamicJsonBuffer size: 73884

It seems that the ESP32 required more memory than for the ESP8266 so I add a little more for each buffer compared than what the Json Assistant gives.

Would you know why is that happening? Is it something with the clear() function I should use?

Thanks a lot for your help!!!

bblanchon, 2018-01-14:
Hi,

This sound like a memory issue.
The execution order matters because it depends on how the heap is fragmented.

Do you keep all the `JsonBuffer` in memory at the same time?
Can you check `JsonBuffer::size()` to make sure that it doesn't grow?

Regards,
Benoit

PS: In [Mastering ArduinoJson](https://arduinojson.org/book/), I show how to parse Weather Underground forecast with less than 2KB.

IoTAll, 2018-01-14:
Hi,
Thank you for your answer! I really appreciate the time you spend for helping others!

In each functions, I am using the same name for the buffer, I only change the size. Like this below. Is it wrong?
```
  const size_t capacity = XXXXX;
  DynamicJsonBuffer jsonBuffer(capacity);

  // Parse JSON object
  JsonObject& root = jsonBuffer.parseObject(client);
```
I used the jsonBuffer.size() function at the end of each of those functions and the json buffer size seems to change a little up and down between each function calls, but not growing, it seems stable.

Doing some more tests, if I call the function successively in this order below, the hourly function parse successfully only the first time, I get the parsing fail from the second time.
1) conditions function
2) forecast function
3) hourly function
4) GoogleMAP API function

Using only the first 3 ones, (not using the GoogleMap API function), it works great wihtout issues.

I put the code for one of those 4 functions below. I built all those functions the same way, only changing the http address request and buffer size. 

I think I am following your advices in the page: "The first parsing succeeds, why do the next ones fail?"

Sorry for the basic questions, I consider myself still as a beginner, especially for those memory topic!

What is the best order for calling each functions? Does it depend on the buffer size?

Do you see anything wrong in my function below?

Thanks a lot for the help!!

```
void WundergroundConditions() {
  char* servername ="api.wunderground.com"; 
  WiFiClient client;
  const int httpPort = 80;
  
  if (!client.connect(servername, httpPort)) {
        client.stop();
        return;
    }
    
      String url;
      url = "/api/"+APIKEY+"/conditions/q/" + GoeCoordinate + ".json";

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + servername + "\r\n" +
                 "Connection: close\r\n\r\n");

    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            client.stop();
            Serial.println(F("Client timeout"));
            return;
        }
    }

  // Check HTTP status
  char status[32] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
  if (strcmp(status, "HTTP/1.1 200 OK") != 0) {
    Serial.print(F("Unexpected response"));
    Serial.println(status);
    client.stop();
    return;
  }

  // Skip HTTP headers
  char endOfHeaders[] = "\r\n\r\n";
  if (!client.find(endOfHeaders)) {
    Serial.println(F("Invalid response"));
    client.stop();
    return;
  }

  // Allocate JsonBuffer
  const size_t capacity = 4000;
  DynamicJsonBuffer jsonBuffer(capacity);

  // Parse JSON object
  JsonObject& root = jsonBuffer.parseObject(client);

if (!root.success())
{
  Serial.println(F("WeatherDataWunderground Conditions parseObject() failed"));
}
else{
  Serial.println(F("WeatherDataWunderground Conditions parseObject() success!"));

  String cityCountry = root["current_observation"]["display_location"]["full"];
  String currentWeatherConditions = root["current_observation"]["weather"];
  String currentWeatherConditionsIcon = root["current_observation"]["icon"];
  String currentWeatherTemp = root["current_observation"]["temp_c"];

}

  // Disconnect
  client.stop();

  Serial.print("JSON BUFFER SIZE: "); Serial.println(jsonBuffer.size());

}
```

bblanchon, 2018-01-15:
Except using `String` excessively, the function above looks correct:

- [X] the `DynamicJsonBuffer` is scoped
- [X] the capacity is passed to the constructor
- [X] the `Stream` is directly passed to `parseObject()`

I don't see any problem regarding the usage of ArduinoJson.

However, I still think the problem is related to heap fragmentation.
If I were you, I would eliminate as much `String` as possible.

If this doesn't solve the problem, you should try to parse the hourly forecast in chunks: one hour after the other.
It requires a bit more code, but it's much more efficient in term of memory usage.
(See the "Case Studies" in  [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues) for a complete walk-through).

bblanchon, 2018-01-22:
@IoTAll, any progress?

IoTAll, 2018-01-22:
Hi!
Thanks for asking!
I actually bought your guide online after you answered me and I am trying to follow your case studies.

I was wondering one thing:

As you show in your case studies 7.2 and 7.3, I use the `wificlient client`, send the http request, then check that I get the answer `"HTTP/1.1 200 OK"` at the beginning of the answer, then remove the header from the answer, and jump to the start of the json file I am interested in thanks to `"client.find"`. At this point I am at the beginning of the first weather hourly forecast object.

Then in your example you use:
`  JsonObject &obj = jb.parseObject(response);`
if I understand correctly, this will parse only one single object you are at with your `client`. Is it right? Or does it parse the entire remaining part of the json file?

What is the difference with this one: I saw it in other examples.
`  JsonObject& root = jsonBuffer.parseObject(client);`
When I use this one, it parses the entire remaining of the json file in the jsonBuffer.

Is it the difference between those 2 lines of code? One parse only the following object you are at with your `client`, and the other one parse the entire remaining json file?

Thanks a lot!

bblanchon, 2018-01-23:
Hi!

Thanks for buying the [book](https://arduinojson.org/book/?utm_source=github&utm_medium=issues), I hope you'll like it.

There is no difference between:

```c++
JsonObject &obj = jb.parseObject(response);
```

and 

```c++
JsonObject& root = jsonBuffer.parseObject(client);
```

They both call [`parseObject(Stream&)`](https://arduinojson.org/api/jsonbuffer/parseobject/?utm_source=github&utm_medium=issues) and they both stop at the closing brace (`}`).

Regards,
Benoit

IoTAll, 2018-01-23:
Ok!
So with the hourly forecast for instance, having 30+ objects, is there a way to parse into the jsonBuffer one forecast object at a time? And not the whole remaining part of the Jason file?

In that case the dynamic memory will need to be at the size of one object only which would be super.

Thanks a lot!

bblanchon, 2018-01-24:
Yes @IoTAll.

I thought that was clear from my previous message: [`parseObject(Stream&)`](https://arduinojson.org/api/jsonbuffer/parseobject/?utm_source=github&utm_medium=issues) always parse one object and stops.

You can apply this feature to parse one forecast at a time.
That is exactly what the [book](https://arduinojson.org/book/?utm_source=github&utm_medium=issues) shows in the case study, except it's using daily instead of hourly.

Regards,
Benoit

IoTAll, 2018-01-26:
Hi!
Sorry my bad, I just forgot the entire json file was also starting with the "{".... Not only the objects.

Works perfect now with low memory usage by parsing the objects one at a time!

Thanks a lot!

bblanchon, 2018-01-29:
You are very welcome.
Thank you for using ArduinoJson.

bblanchon, 2018-01-29:
PS: don't forget to add a ⭐️ to help the project ;-)

closed 2018-01-29 by bblanchon

IoTAll, 2018-01-29:
done!

HanyTalaatRamadan, 2018-03-05:
Hi , 
Thanks a lot for the great Library , it was very helpful for me, and i'm new IoT
i'm using NodeMCU v1 and I'm using ArduinoJson to parse JSON response from REST web service , the Web Service response contains multiple Objects in JSON as following 

> {
>     "items" : [
>         {
>             "ThingId" : 2,
>             "ThingName" : "Tank-CAT3406-1",
>             "PropertyName" : "Radius",
>             "PropertyValue" : "3.5",
>             "DeviceId" : 5,
>             "SensorId" : 2,
>             "ThingTypeId" : 2,
>             "Recid" : 1,
>             "links" : [
>                 {
>                     "rel" : "self",
>                     "name" : "ThingPropertiesRO1",
>                     "kind" : "item"
>                 },
>                 {
>                     "rel" : "canonical",
>                     "name" : "ThingPropertiesRO1",
>                     "kind" : "item"
>                 }
>             ]
>         },
>         {
>             "ThingId" : 2,
>             "ThingName" : "Tank-CAT3406-1",
>             "PropertyName" : "Width",
>             "PropertyValue" : "10",
>             "DeviceId" : 5,
>             "SensorId" : 2,
>             "ThingTypeId" : 2,
>             "Recid" : 2,
>             "links" : [
>                 {
>                     "rel" : "self",
>                     "name" : "ThingPropertiesRO1",
>                     "kind" : "item"
>                 },
>                 {
>                     "rel" : "canonical",
>                     "name" : "ThingPropertiesRO1",
>                     "kind" : "item"
>                 }
>             ]
>         },
>         {
>             "ThingId" : 2,
>             "ThingName" : "Tank-CAT3406-1",
>             "PropertyName" : "Hight",
>             "PropertyValue" : "5",
>             "DeviceId" : 5,
>             "SensorId" : 2,
>             "ThingTypeId" : 2,
>             "Recid" : 3,
>             "links" : [
>                 {
>                     "rel" : "self",
>                     "name" : "ThingPropertiesRO1",
>                     "kind" : "item"
>                 },
>                 {
>                     "rel" : "canonical",
>                     "name" : "ThingPropertiesRO1",
>                     "kind" : "item"
>                 }
>             ]
>         }
>     ],
>     "count" : 3,
>     "hasMore" : false,
>     "limit" : 5,
>     "offset" : 0,
>     "links" : [
>         {
>             "rel" : "self",
>             "href" : "http://127.0.0.1:7101/PowerSenseApp-RESTWebService-context-root/rest/beta/ThingPropertiesRO1",
>             "name" : "ThingPropertiesRO1",
>             "kind" : "collection"
>         }
>     ]
> }

when i'm trying to parse the  response to multiple Json Objects it fails and return null , but for one object in response it works fine 

```
void getTankInit(String stringResponse){
  const size_t bufferSize = 16580;//JSON_ARRAY_SIZE(1) + 3*JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + 7*JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(6) + 3*JSON_OBJECT_SIZE(9) + 1748000;
  Serial.println(bufferSize);
  Serial.println(stringResponse.length());
DynamicJsonBuffer jsonBuffer(bufferSize);
JsonObject& root = jsonBuffer.parseObject(stringResponse);
Serial.println(root.size());
JsonArray& items = root["items"];
for (int i=0; i<items.size());i++) {
JsonObject& items0 = items[i];
const char* items0_ThingName = items0["ThingName"]; 
const char* items0_PropertyName = items0["PropertyName"]; 
const char* items0_PropertyValue = items0["PropertyValue"]; 

Serial.println(items0_PropertyName); Serial.println(items0_PropertyValue);
}
```
i appreciate your help for my case .