bblanchon/ArduinoJson #667: Proper assignment usage of "set" with signed char?

Humancell, 2018-01-18:
Hello,

As usual, this is a great library!  :-)

I'm using this in the Arduino environment.  My issue relates to having a "signed char" appear in the JSON properly as a numeric value from -128 to 127.

I have a change that I made in my code, and am now seeing an error in what I have done.  I want to ensure that I'm doing this properly ... I'm sure it is me.

I was using an "unsigned byte" for a value, and things were working ok:

```
  StaticJsonBuffer<300> jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
  json["unsigned_byte"] = myUnsignedValue;    // byte myUnsignedValue

```
Then I needed to change the unsigned byte to a signed "char" as we needed -128 to 127 ... and things broke immediately.

I left the code, but this did not work:

```
  StaticJsonBuffer<300> jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
  json["signed_char"] = myChar;    // char myChar

```
So then I tried:

```
  StaticJsonBuffer<300> jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
  json["signed_char"] = int(myChar);    // char myChar

```
Is this where I should be using "set()"?

```
  StaticJsonBuffer<300> jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
  json.set("signed_char", myChar);    // char myChar

```
What I was hoping to see is:  {"signed_char" : -34}

But I get non-printable chars.

I read the FAQ where it references the "garbage" in the JSON, which is what I'm seeing (or non-printable characters) and realize that I'm not doing this correctly ... even though "byte" worked ok.

bblanchon, 2018-01-19:
Hi,

> I left the code, but this did not work:

I cannot reproduce, see: https://wandbox.org/permlink/3VxCojhElwsfT34h

> But I get non-printable chars.

I cannot reproduce either, see: https://wandbox.org/permlink/egyoLCETsnrq4SJ1

Regards,
Benoit

bblanchon, 2018-01-26:
@Humancell, any progress?

closed 2018-02-02 by bblanchon

Humancell, 2018-02-03:
I'm sorry I didn't reply sooner ... after your last inquiry I checked with testing, and we found our bug.  There were two code paths, and only one of them had been updated.  It's working correctly!

Thank you for you quick response ... sorry I was slow to reply.