bblanchon/ArduinoJson #668: StaticJsonBuffer and DynamicJsonBuffer size limited to 256 bytes compiled with chipkitCore 1.4.3 for Arduino IDE?

yoverload, 2018-01-18:
Hi, I am trying to parse string witch is over 2k.  For Arduino Mega2560 Board  works fine. 
With strings shorter than 256 Bytes , it works for ChipKit Board MAX32. Anything bigger , complies but does not work.
I am using modified "StringExample" from examples included in the library. 
Here is the part that is not from example: 

```
const size_t bufferSize = 2*JSON_ARRAY_SIZE(2) + 48*JSON_ARRAY_SIZE(4) + JSON_ARRAY_SIZE(6) + JSON_ARRAY_SIZE(7) + 2*JSON_ARRAY_SIZE(24) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(3);
DynamicJsonBuffer jsonBuffer(bufferSize);
```
That is the suggested buffer size for the given json string.
```
{
"day1":{
	"GroupsCount":"2",
	"lampAddsByGroup":[
					["5","1","2","3","4","5"],
					["6","6","7","8","9","10","11"]
				],
	"transitions":[
				[
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"]
				],
				[
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"],
					["1200","1300","100","100"]
				]
			]
	}
}
```
The following code
```
JsonObject& root = jsonBuffer.parseObject(String(jsonTest));
String output;
root.printTo(output);
Serial.println(output);
```
gives output on serial monitor using chipKit MAX32 Borad:
`{}`

When the string size is reduced to 256 Bytes , it woks.

bblanchon, 2018-01-19:
Hi @yoverload,

Many platforms limit the size of the stack, that why you cannot use a big `StaticJsonBuffer`.
You just need to switch to a `DynamicJsonBuffer` and it should be fine.

BTW, I noticed that you use `String` extensively in your program.
I strongly recommend against this practice because it causes a lot of duplications and wastes a lot of memory.

Regards,
Benoit

yoverload, 2018-01-19:
@bblanchon  Thanks for your response..

> You just need to switch to a DynamicJsonBuffer and it should be fine.

`DynamicJsonBuffer`  does not work as well.. 

> BTW, I noticed that you use String extensively in your program.
> I strongly recommend against this practice because it causes a lot of duplications and wastes a lot of memory.

-----The platform i'm using  come with 128KB of RAM.  Using char array does not help. 

There is something to do with configuration of chipkit core. For Arduino boards works as described , using Strings or char array. 

Regards,
Jordan

bblanchon, 2018-01-22:
Hi @yoverload,

How can I help you?

Regards,
Benoit

closed 2018-01-29 by bblanchon