bblanchon/ArduinoJson #669: "Redefined" errors when compiling with CCS (TI)

adamjones1, 2018-01-22:
First, I'm trying to swap ArduinoJson into a semi-mature product that uses another json library - so these errors are new.

I have a project where I must include ArduinoJson.h in many files (there is a serial receiver that does checks what is received, then figures out which command is being called and then passes the reference to the root object off to the appropriate handler for that specific class). However, when I try to use any of the same functionality from the library in multiple files, I get errors like this:

error #10056: symbol "initialization guard variable for ArduinoJson::JsonObject::invalid()::instance (instance 0)" redefined: first defined in "./common/em_main.obj"; redefined in "./common/serial/serial_link.obj"
error #10056: symbol "initialization guard variable for ArduinoJson::JsonArray::invalid()::instance (instance 0)" redefined: first defined in "./common/em_main.obj"; redefined in "./common/serial/serial_link.obj"

In this case, all em_main contains is the following:
DynamicJsonBuffer jsonBuffer;
char *test = "{"abc":123}";
JsonObject &root = jsonBuffer.parse(test);

I'm curious what could be happening here. Furthermore, since I haven't used a header-only library before, I seriously wonder if the code will be duplicated all over the place, since the TI complier says that it doesn't do link-time optimization for files with templates.

bblanchon, 2018-01-22:
Hi @adamjones1,

Assuming TI is using GCC, you can try to add the flags `-fno-threadsafe-statics`.
It should remove the "initialization guard variable" that causes the link error.

This is the first time I see that.
I have several programs using ArduinoJson in different compilation units and it never caused a link problem.
By the way, don't worry about duplicated code, the linker keeps only one copy of a function, this doesn't require LTO.

Regards,
Benoit

adamjones1, 2018-01-23:
Thank you, Benoit. Unfortunately, although the ArduinoJson interface is quite nice, I think that I'll have to put off swapping the libraries for later. Therefore, I won't be able to immediately verify this. However, hopefully your response helps someone else as well.

closed 2018-01-23 by adamjones1

bblanchon, 2018-01-23:
Hi @adamjones1,

I'm sorry to hear that.
I hope you'll have other opportunities to use ArduinoJson.

For the record, I plan on removing the static variable `ArduinoJson::JsonObject::invalid()::instance` in version 6, as part of #309.

Regards,
Benoit