bblanchon/ArduinoJson #670: Same of issue 118 but conversion from byte

dariocdj, 2018-01-25:
Hello,
I've : 
root["config.IP"] = config.IP;

and  config.IP[0] = IPABYTE_1;config.IP[1] = IPABYTE_2;config.IP[2] = IPABYTE_3;config.IP[3] = IPABYTE_4;

This is error :
error: no match for 'operator=' (operand types are 'ArduinoJson::JsonObjectSubscript<const char*>' and 'byte [4] {aka unsigned char [4]}')

  root["config.IP"] = config.IP;

Thanks
Dario Cdj

dariocdj, 2018-01-25:
if i try to use : reinterpret_cast<const char*>(config.IP) : 

"config.IP":"À¨yÿÿÿ"

also tried as.String ....

bblanchon, 2018-01-26:
Hi @dariocdj,

It seems that you want to convert a 4-byte array into a string representing an IPv4 address.
The proper way to do that is:

```c++
char ip[16];
sprintf(ip, "%d.%d.%d.%d", IPABYTE_1, IPABYTE_2, IPABYTE_3, IPABYTE_4);

root["config.IP"] = ip;
```

Regards,
Benoit

PS: If you need an explanation on how strings work in C and C++, have a look at "The missing C++ course" in [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues).

closed 2018-02-02 by bblanchon