bblanchon/ArduinoJson #673: Parsing Json from Sd card on arduino Yun

Shinyhero36, 2018-02-04:
Hello,
here is my json file:
`{
  "data": [
    {
      "nom": "leban",
      "prenom": "gerald",
      "jetons": "10",
      "numero": "9562"
    },
    {
      "nom": "billon",
      "prenom": "yannis",
      "jetons": "10",
      "numero": "4562"
    },
    {
      "nom": "valencia",
      "prenom": "gael",
      "jetons": "10",
      "numero": "7412"
    },
    {
      "nom": "bru",
      "prenom": "joris",
      "jetons": "10",
      "numero": "8426"
    },
    {
      "nom": "monteleone",
      "prenom": "guillaume",
      "jetons": "10",
      "numero": "1256"
    }
  ]
}`
I want to parse it with arduino without adding this king of line:
const char* json = "{\"data\":[{\"nom\":\"leban\",\"prenom\":\"gerald\",\"jetons\":\"10\",\"numero\":\"9562\"},{\"nom\":\"billon\",\"prenom\":\"yannis\",\"jetons\":\"10\",\"numero\":\"4562\"},{\"nom\":\"valencia\",\"prenom\":\"gael\",\"jetons\":\"10\",\"numero\":\"7412\"},{\"nom\":\"bru\",\"prenom\":\"joris\",\"jetons\":\"10\",\"numero\":\"8426\"},{\"nom\":\"monteleone\",\"prenom\":\"guillaume\",\"jetons\":\"10\",\"numero\":\"1256\"}]}";

how can i do?

bblanchon, 2018-02-05:
Hi @Shinyhero36,

Please have a look at the example [JsonConfigFile.ino](https://arduinojson.org/example/config/?utm_source=github&utm_medium=issues)

Regards,
Benoit

bblanchon, 2018-02-06:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)

closed 2018-02-06 by bblanchon