bblanchon/ArduinoJson #675: Using default value problem (x=json["Test"] | x;)

ttlappalainen, 2018-02-05:
I have problem with using default value. I have on json: "Test": 1
```
  const char *strTest;
  double v=5;
  v=jsonProperty[strTest] | v;

```
Works fine and returns 1, but
```
  uint8_t v=5;
  v=jsonProperty[strTest] | v;

```
returns 5 allways. Last works fine without | v

bblanchon, 2018-02-05:
Hi @ttlappalainen,

I cannot reproduce the bug.
See demo: https://wandbox.org/permlink/Q0FLElI8oJ9nuThD

Please provide an [MCVE](https://stackoverflow.com/help/mcve).

Regards,
Benoit

ttlappalainen, 2018-02-06:
Now I am not sure is this bug or feature. With below code first returns wrong and second right. Also I expect that the reason is that you when I am using existing data, it has internally double object.

```
#include <Arduino.h>
#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200);
  delay(300);

  Serial.println("Startup");

  DynamicJsonBuffer jsonBuffer(1000);
  jsonBuffer.clear();
  JsonObject& jsonControl=jsonBuffer.createObject();
  double a=100;
  jsonControl["Test"]=a;

  jsonControl.prettyPrintTo(Serial);

  uint8_t v=jsonControl["Test"] | 5;
  
  Serial.println();
  Serial.print("Read value:"); Serial.println(v);

  char buf[1000];
  jsonControl.prettyPrintTo(buf,1000);
  DynamicJsonBuffer jsonBuffer2(1000);
  JsonObject& jsonControl2=jsonBuffer2.parseObject(buf);
  v=jsonControl2["Test"] | v;
  Serial.println();
  Serial.print("Read value:"); Serial.println(v);
}

void loop() {
}

```

bblanchon, 2018-02-07:
```c++
DynamicJsonBuffer jsonBuffer(1000);
char buf[1000];
DynamicJsonBuffer jsonBuffer2(1000);
```

Are you sure that there is enough RAM on your micro-controller?

ttlappalainen, 2018-02-07:
Teensy 3.5 RAM 196 kB. Totally with flash more than on my first computer:)

bblanchon, 2018-02-08:
OK, now I see it.

`jsonControl["Test"]` is a `double` so when you try to extract a `uint8_t`, it returns the default value.

To me, it's a bug because it's not consistent with:

```c++
uint8_t v = jsonControl2["Test"]; // return 100 as expected
```

ttlappalainen, 2018-02-08:
OK - thanks. Meanwhile I can go around problem with
`jsonControl["Test"]=(uint8_t)a;`
where it is possible.

closed 2018-02-09 by bblanchon