bblanchon/ArduinoJson #680: Cant parse HTTP response(Static String works)

KowalikDominik, 2018-02-14:
I have a problem with parsing String receive json from esp8266 module.
Arduino code:
```
#include <SimpleDHT.h>

#include <WiFiEsp.h>
#include <WiFiEspUdp.h>
#include <ArduinoJson.h>

#ifndef HAVE_HWSERIAL1
#include "SoftwareSerial.h"
SoftwareSerial Serial1(3, 2); // RX, TX
#endif

char ssid[] = "NETIASPOT";            // your network SSID (name)
char pass[] = "";        // your network password
int status = WL_IDLE_STATUS;     // the Wifi radio's status

char server[] = "192.168.1.2";

// config temp & hum sensors
int dataPinSensor1 = 13;
int dataPinSensor2 = 12;
SimpleDHT11 dht11;
SimpleDHT22 dht22;
byte temperatureIn, humidityIn;
byte temperatureOut, humidityOut;
//
//config analog input from soil hum sensor
int sensorPin = A0; // select the input pin for the potentiometer
int sensorValue = 0;
float aReading = 1023.0;// max input
float aLoad = 0.0; //
float bReading = 500.0;// min input
float bLoad = 100.0; //
float load;
//

//Config relay to control valve
const int valve1 = 10;
const int valve2 = 11;
int state1,state2;
//

String DataToSend;

 
unsigned long lastConnectionTime = 0;         // last time you connected to the server, in milliseconds
const unsigned long postingInterval = 3000L; // delay between updates, in milliseconds

// Initialize the Ethernet client object
WiFiEspClient client;

void setup()
{
  //config valve pin
  pinMode(valve1, OUTPUT);
  pinMode(valve2, OUTPUT);
  digitalWrite(valve1, HIGH);
  digitalWrite(valve2, HIGH);

 
  
  // initialize serial for debugging
  Serial.begin(115200);
  // initialize serial for ESP module
  Serial1.begin(9600);
  // initialize ESP module
  WiFi.init(&Serial1);

  // check for the presence of the shield
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue
    while (true);
  }

  // attempt to connect to WiFi network
  while ( status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WPA SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network
    status = WiFi.begin(ssid, pass);
  }

  Serial.println("You're connected to the network");
  
  printWifiStatus();
}

void loop()
{
  String json;

  int lines_received = 0;
  
  while(client.available()) {
    String line = client.readStringUntil('\r\n');
    if (lines_received == 9) { 
      json=line;  
    }
    lines_received++;
  }
  
  if(json.length()>0){
    Serial.println("from server: "+json);
  
    //String json1 ="{\"id\":3,\"data\":[{\"start\":\"11:12:23\",\"stop\":\"13:11:11\"},{\"start\":\"11\",\"stop\":\"13\"}]}";
  
    const size_t bufferSize = JSON_ARRAY_SIZE(2) + 3*JSON_OBJECT_SIZE(2);
    DynamicJsonBuffer jsonBuffer(bufferSize);
    JsonObject& root = jsonBuffer.parseObject(json1);
  
    root.prettyPrintTo(Serial);
  
  
  }
  if (millis() - lastConnectionTime > postingInterval) {
    httpRequest();
  }
   
 
}

// this method makes a HTTP connection to the server
void httpRequest()
{
  Serial.println();
    
  // close any connection before send a new request
  // this will free the socket on the WiFi shield
  client.stop();

  // if there's a successful connection
  if (client.connect(server, 80)) {
    Serial.println("Connecting...");
    
    // send the HTTP PUT request
   // String DataToSend = "?id=1&pol=100"; 
    client.println("GET /woda/tester.php HTTP/1.1");
    client.println(F("Host: 192.168.1.2"));
    client.println("Connection: close");
    client.println();
    // note the time that the connection was made
    lastConnectionTime = millis();
  }
  else {
    // if you couldn't make a connection
    Serial.println("Connection failed");
  }
}
```

Result in serial monitor:

```
[WiFiEsp] Connecting to 192.168.1.2
Connecting...
from server: "{\"id\":3,\"data\":[{\"start\":\"11:12:23\",\"stop\":\"13:11:11\"},{\"start\":\"11\",\"stop\":\"13\"}]}"
{}
[WiFiEsp] Connecting to 192.168.1.2
Connecting...
from server: "{\"id\":3,\"data\":[{\"start\":\"11:12:23\",\"stop\":\"13:11:11\"},{\"start\":\"11\",\"stop\":\"13\"}]}"
{} 

```

When I uncomment this line parser works:
`//String json1 ="{\"id\":3,\"data\":[{\"start\":\"11:12:23\",\"stop\":\"13:11:11\"},{\"start\":\"11\",\"stop\":\"13\"}]}";`

Result in serial:

```
[WiFiEsp] Connecting to 192.168.1.2
Connecting...
from server: "{\"id\":3,\"data\":[{\"start\":\"11:12:23\",\"stop\":\"13:11:11\"},{\"start\":\"11\",\"stop\":\"13\"}]}"
{
  "id": 3,
  "data": [
    {
      "start": "11:12:23",
      "stop": "13:11:11"
    },
    {
      "start": "11",
      "stop": "13"
    }
  ]
} 
```
I have no idea. What is wrong?

bblanchon, 2018-02-15:
Hi @KowalikDominik,

I'm very surprised that the string coming from the server is escaped.

It should not be:

    from server: "{\"id\":3,\"data\":[{\"start\":\"11:12:23\",\"stop\":\"13:11:11\"},{\"start\":\"11\",\"stop\":\"13\"}]}"

Instead, it should be:

    from server: {"id":3,"data":[{"start":"11:12:23","stop":"13:11:11"},{"start":"11","stop":"13"}]}

Please check the server code.

Regards,
Benoit

KowalikDominik, 2018-02-15:
Hi @bblanchon 

I'm ashamed. I made a fundamential error. Thank you for your reply.
In PHP Server i wrote this code:

`
echo json_encode($response); 
`

and i throught everything was fine and wasnt looking for error here.
I change Php code to:

`
echo $response;
 `

And now everything works.

bblanchon, 2018-02-16:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)

closed 2018-02-16 by bblanchon