bblanchon/ArduinoJson #684: Unable to read headers and parse

mars000, 2018-02-22:
Hi

I am using latest V5.13.1 library and V2.4.0 core for Arduino ESP8266 using wemos mini Pro board.

I have two sketches. Sketch 1 below works ok. Sketch 2 does not, I get "no response or invalid response" when trying to "skipHeaders". I can't seem to work out why.

Sketch 1 [this works]
```cpp
#define BLYNK_PRINT Serial

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "xxxxx";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "xxx";
char pass[] = "xxx";

void setup()
{

  // Debug console
  Serial.begin(115200);
  WiFi.begin(ssid, pass);           // Connect to WiFi network
  long waittime = millis() + 14000; // 14 seconds wait time

  while (WiFi.status() != WL_CONNECTED)
  { // Wait for board to connect to WiFi network
    delay(500);
    Serial.print(F("."));
    if (millis() > waittime)
    {
      Serial.println(F("breaking !"));
      break;
    }
  }
  Blynk.config(auth, "blynk.xxx.com", 8080);
  Blynk.connect();

  check_firmware_version("blynk.xxxx.com", "/xxxx/csr0002_11_4d_8c_0f.json", 1);
}

int counter = 0;

BLYNK_WRITE(127)
{
  Serial.print(counter++);
  Serial.println(" : Button pressed");
}

void loop()
{
  Blynk.run();
}

#define HTTP_TIMEOUT 200

void check_firmware_version(char *json_server, char *json_resource, int local_CURRENT_VERSION)
{
  WiFiClient client_json;
  // firmware_check return_result;
  // return_result.error_detected = false;
  if (connect(json_server, &client_json))
  {
    // sendRequest()
    Serial.print(F("GET "));
    Serial.println(json_resource);

    client_json.print("GET ");
    client_json.print(json_resource);
    client_json.println(" HTTP/1.0");
    client_json.print("Host: ");
    client_json.println(json_server);
    client_json.println("Connection: close");
    client_json.println();
    //------

    //skipResponseHeaders()
    char endOfHeaders[] = "\r\n\r\n";

    client_json.setTimeout(HTTP_TIMEOUT);
    bool ok = client_json.find(endOfHeaders);

    if (!ok)
    {
      Serial.println(F("No response or invalid response!"));
    }

    Serial.println(F("I'm just before client_json.available()"));
    //--------

    //	json_file_data jsondata;
    DynamicJsonBuffer jsonBuffer;
    // readResponseContent()
    JsonObject &root = jsonBuffer.parseObject(client_json); 

    Serial.println("after parsing function called");

    if (!root.success())
    {

      Serial.println(F("JSON parsing failed!"));

     
    }
  }

  Serial.println(F("Parsing successful !"));
}
//------

bool connect(const char *hostName, WiFiClient *client_json)
{
	Serial.print(F("Connect to "));
	Serial.println(hostName);

	bool ok = client_json->connect(hostName, 80);

	Serial.println(ok ? "Connected" : "Connection Failed!");
	return ok;
}

```

SKETCH 2 below [THIS DOES NOT WORK]

```
cpp
#define BLYNK_PRINT Serial

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>

struct children_object
{
	char appliance[23];
	char mac_address[18]; // 17 required for code
	char auth_code[32];   // Note: larger than 32 as it needs to hold original Encrypted verison

};

struct json_file_data
{
	int fileversion;
	char filename[32];
	char download_URL[70];
	char date_created[16];
	char hardware[16];
	char blynk_server[32];
	char auth_code[32];
	short int total_children;
	short int total_peers; // new
	int blynk_https_port;

	children_object children[9];

	char parent_appliance[23];
	char parent_mac_address[12];
	char parent_auth_code[32];
};

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "xxxx";

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "xxxx";
char pass[] = "xxxx";

void setup()
{

  // Debug console
  Serial.begin(115200);
  WiFi.begin(ssid, pass);           // Connect to WiFi network
  long waittime = millis() + 14000; // 14 seconds wait time

  while (WiFi.status() != WL_CONNECTED)
  { // Wait for board to connect to WiFi network
    delay(500);
    Serial.print(F("."));
    if (millis() > waittime)
    {
      Serial.println(F("breaking !"));
      break;
    }
  }
  Blynk.config(auth, "blynk.xxxx.com", 8080);
  Blynk.connect();

  check_firmware_version("blynk.xxxx.com", "/csr0002_11_4d_8c_0f.json", 1);
}

int counter = 0;

BLYNK_WRITE(127)
{
  Serial.print(counter++);
  Serial.println(" : Button pressed");
}

void loop()
{
  Blynk.run();
  // You can inject your own code or combine it with other sketches.
  // Check other examples on how to communicate with Blynk. Remember
  // to avoid delay() function!
}

#define HTTP_TIMEOUT 20

void check_firmware_version(char *json_server, char *json_resource, int local_CURRENT_VERSION)
{
	WiFiClient client_json;
	if (connect(json_server, &client_json))
	{ // if1
	if (sendRequest(json_server, json_resource, &client_json) && skipResponseHeaders(&client_json))
		{ // if2
			
		  json_file_data jsondata;

			if (readResponseContent(&jsondata, &client_json)) // was &client_json
			{ // if3

				Serial.print(F("Current Firmware Version = "));
      }
    }
  }
}

bool connect(const char *hostName, WiFiClient *client_json)
{
	Serial.print(F("Connect to "));
	Serial.println(hostName);

	bool ok = client_json->connect(hostName, 80);

	Serial.println(ok ? "Connected" : "Connection Failed!");
	return ok;
}

bool sendRequest(const char *host, const char *resource, WiFiClient *client_json)
{
	Serial.print(F("GET "));
	Serial.println(resource);

	client_json->print("GET ");
	client_json->print(resource);
	client_json->println(" HTTP/1.0");
	client_json->print("Host: ");
	client_json->println(host);
	client_json->println("Connection: close");
	client_json->println();

	return true;
}

// Skip HTTP headers so that we are at the beginning of the response's body

bool skipResponseHeaders(WiFiClient *client_json)
{
	// HTTP headers end with an empty line
	// bool ok = true;
	char endOfHeaders[] = "\r\n\r\n";

	client_json->setTimeout(HTTP_TIMEOUT);
	bool ok = client_json->find(endOfHeaders);

	if (!ok)
	{
		Serial.println(F("No response or invalid response!"));
	}

	Serial.println(F("I'm just before client_json.available()"));

	return ok;
}

bool readResponseContent(struct json_file_data *jsondata, WiFiClient *client_json)
{
	// Compute optimal size of the JSON buffer according to what we need to parse.
	// This is only required if you use StaticJsonBuffer.

	DynamicJsonBuffer jsonBuffer;

	Serial.println(F("waiting for parsing to occur..."));
	JsonObject &root = jsonBuffer.parseObject(*client_json);  // was *client_json

	if (!root.success())
	{

		Serial.println(F("JSON parsing failed!"));

		return false;
	}
 else
 {
   Serial.println(F("Parsing successful !"));
 }

	
	// Here were copy the strings we're interested in
}

```

```
No response or invalid response!
I'm just before client_json.available()
```

bblanchon, 2018-02-23:
Hi @mars000,

I looked carefully at the code sample, and it seems correct.
Can you try to reduce the amount of code until you find the single change that breaks the program?

Regards,
Benoit

mars000, 2018-02-23:
hi Benoit
it seems as though its a memory issue. I reduced my program stack usage and seems to work ok now.

bblanchon, 2018-02-28:
Excellent!

closed 2018-02-28 by bblanchon