bblanchon/ArduinoJson #687: Support for parsing ISO8601 time string?

huming2207, 2018-02-22:
Hi,

I'm writing a program which needs to convert ISO8601 time string (e.g. `2018-02-22T13:21:00Z` as general or `20180222T132100Z` for some special cases) to `time_t` or `struct std::tm`.  

Currently I'm using `strptime()` function to parse the time string to those two time data types. I'm wondering if I can use `.as<time_t>()` or `.as<struct std::tm>()`? 

Regards,
Jackson

bblanchon, 2018-02-23:
Hi Jackson,

I plan on adding *custom converters* in the future so that anyone can enrich ArduinoJson to support new types.
Unfortunately, I'm currently in the middle of a transformation in the API, so this new feature will have to wait a little.

Regards,
Benoit

huming2207, 2018-02-23:
Hi Benoit,

Thanks for your works, I'm looking forward to try that. 

Regards,
Jackson

Triangulix, 2020-01-26:
Hi 
I would also like to chime in on this. I'm currently looking for a way to have enum values serialized as names and not as numbers. I guess custom converters would be a way to implement this.

closed 2021-03-20 by bblanchon

bblanchon, 2021-05-05:
Feature published in [ArduinoJson 6.18.0](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

commit d7f5b56c (2021-03-20): Added support for custom converters (closes #687)