bblanchon/ArduinoJson #688: Json on Nano

jack4645, 2018-02-25:
I have code that works just fine on the Mega but fails on the Nano. This is a chunk of what I have:
`//------------json parsing -----------------
      unsigned int i = 0; //timeout counter
      char json[buffer] ="{"; // array for Json parsing
      int n = 1;          // character counter for json  
      
      for (int j=0;j<num_elements;j++){
        Serial.print("...Conditon = ");
        Serial.println(conds[j]);
       while (!esp.find(conds[j])){} // find the part we are interested in.
      
        String Str1 = conds[j];
        for (int l=0; l<(Str1.length());l++)
            {json[n] = Str1[l];
             n++;}   
        while (i<60000) {
          if(esp.available()) {
            char c = esp.read();
             
              if(c==',') break;
              json[n]=c;
              n++;
              i=0;
            }
            i++;
          }
         if (j==num_elements-1)
            {json[n]=close_brace;}
         else   
            {json[n]=comma;}
         n++;   
      }
      
     Serial.println("----------json-------------");
     Serial.println(json);

      parseJSON(json);`
And here is the Parse function:
`void parseJSON(char json[300])
    {
     StaticJsonBuffer<buffer> jsonBuffer;
     JsonObject& root = jsonBuffer.parseObject(json);
     
     if (!root.success())
    {
      Serial.print("?fparseObject() failed");
      //return;
    }
    Serial.println("....parseObject() SUCCESS");
    Serial.println(""); 
     const char* city = root["city"];
     const char* weather = root["weather"];
     double temp_f = root["temp_f"];
     const char* humidity = root["relative_humidity"];
     const char* wind_dir = root["wind_dir"];
     double wind_mph = root["wind_mph"];
     const char* icon = root["icon"];
     
     
     // location of conditions
     Serial.print("City    ");
     Serial.println(city);
     
     // Conditions: Sunny, Cloudy, Fog, Rain, etc. 
     Serial.print("Current Conditons  ");
     Serial.println(weather);
      

      
    } //End of parseJSON`
Any idea of what is wrong?

bblanchon, 2018-02-26:
Hi @jack4645,

The Mega has 4 times more RAM than the Nano, so your problem is probably related to memory usage.

The Nano has only 2 KB of RAM, so you need to be very careful not to waste a single byte.
In particular, don't use dynamic allocation at all.
I see that you use a `StaticJsonBuffer`, which is good; but you're also using a `String`.

Regards,
Benoit

jack4645, 2018-02-26:
I was noticing a change in what would happen when I changed the value of the buffer. I will work on the code and try and get the footprint down.
Thanks for your help.

closed 2018-02-26 by jack4645