bblanchon/ArduinoJson #689: Global variable

luigi67, 2018-02-27:
I would use thr ArduinoJson library in a project where an Arduino yun sends the values of some sensors. below  the raw structure

SAcqua                        True||False
SGas                            True||False
SFiamma                     True||False
STemp1T                    could be the temperature or 99 to indicate read error 
STemp1U                    could be the humidity or 99 to indicate read error
STemp2T                    could be the temperature or 99 to indicate read error 
STemp2U                    could be the humidity or 99 to indicate read error
RAcqua                        It is a relay and value could be
                                      10 auto/disabled
                                      11 auto/enabled
                                      20 manual/disabled
                                      21 manual/enabled 
RMotori                        as RAcqua
RGas                            as RAcqua
Mail                               0||1||3 

To make a fat-free document i think to replace the key (eg. SAcqua,STemp2T...) with a char[2] ("0","2"..."10")

The result should be

{
"0": "False",
"1":"False",
"2": "False",
"3": 25,
"4": 60,
"5": 22,
"6": 55,
"7": 10,
"8": 10,
"9": 10,
"10": 0
}

I think to replace the string False||True  with 0||1, but this  is not  the question. I read the book Mastering ArduinoJson quickly, I will read again, maybe I not understood the correct  use of this library, beecause I would use the JSON document above with a StaticJsonBuffer allocated only one time on the setup() function, because what will changes in my program will be the values, not the key, but in the book the use of the buffer as global it is indicated as incorrect.

In particular I would use the JSON document above only to send a reply, while the request should be:

{
"10": 1
}

to request sensors status. Or:

{
"7": 10,
"8": 10,
"9": 10,
"10": 2
}

to modify  the relays'status.

My quenstion is about the correct use of JSON document in this scenario, because I think to use a global JSON document.
If isn't correct, to optimize the memory's use is better to use a common array of byte/short ( byte Sensors [10];).  In this case  is better to use a byte or a short to reduce use of memory?

Tanks
Luigi

bblanchon, 2018-02-28:
Hi Luigi,

This sentence surprised me:

> To make a fat-free document i think to replace the key (eg. SAcqua,STemp2T...) with a char[2] ("0","2"..."10")

If this is a concern, why do you want to use JSON in the first place?
Wouldn't it be easier to use a structure like the following?

```c++
struct RAcqua {
  bool is_auto;
  bool is_enabled;
};

struct SensorData {
  bool SAcqua;
  bool SGas;
  bool SFiamma;
  int STemp1T;
  int STemp1U;
  int STemp2T;
  int STemp2U;
  RAcqua RMotori;
  RAcqua RGas;
  int Mail;
};
```

And then send/receive the structure as-is, as a binary protocol.
This solution is fast and small.

Generally, we use JSON when we want a protocol that is either:

* human-readable
* dynamic (messages are different from one another)
* evolutive (we can add members and keep the old ones)

These goals are in contradiction with:

* being small
* being fast

So, I'm not sure JSON is a good fit for your project.

About the global `JsonBuffer`, I can't say this enough: this is always a bad idea.
Your Arduino Yun only has 2.5KB of RAM; you cannot afford to waste a huge part on a JsonObject.
My advice is to keep a custom data structure in memory and serialize it in one shot, as it's shown in:

* [JsonConfigFile.ino](http://arduinojson.org/example/config/?utm_source=github&utm_medium=issues)
* [What's the best way to use the library](http://arduinojson.org/faq/whats-the-best-way-to-use-the-library/?utm_source=github&utm_medium=issues)
* In the case studies of the book

Regards,
Benoit

luigi67, 2018-03-01:
Hi Benoit
Thanks for the reply
You confirmed me what I thought, but to make a really  fat-free document is better a struct as you had descrived or an array? Sorry but I'm not expert about embedded devices

bblanchon, 2018-03-02:
Hi @luigi67,

To make it fat-free and fast, choose a binary protocol.
To make it human-readable, extensible and easy to use with other languages, choose JSON.

Regards,
Benoit

closed 2018-03-09 by bblanchon