bblanchon/ArduinoJson #690: Compilation Error on Arduino

DiE-MaRMeLaDe, 2018-03-04:
When I try to compile included library, I see this:
```
Arduino: 1.8.5 (Mac OS X), Board:"Arduino/Genuino Uno"

In file included from /Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:12:0,
                 from /Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:10,
                 from /Users/mrmld/Projects/MRobotics.Volt/MRobotics.Volt.CryptoContainer/MRobotics.Volt.CryptoContainer.ino:6:
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator ArduinoJson::JsonObject::findKey(TStringRef) [with TStringRef = const int&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator = ArduinoJson::Internals::ListIterator<ArduinoJson::JsonPair>]':
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:303:66:   required from 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::const_iterator ArduinoJson::JsonObject::findKey(TStringRef) const [with TStringRef = const int&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::const_iterator = ArduinoJson::Internals::ListConstIterator<ArduinoJson::JsonPair>]'
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:309:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson::JsonObject::get_impl(TStringRef) const [with TStringRef = const int&; TValue = unsigned char; typename ArduinoJson::Internals::JsonVariantAs<TValue>::type = unsigned char]'
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:170:48:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type ArduinoJson::JsonObject::get(const TString&) const [with TValue = unsigned char; TString = int; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type = unsigned char]'
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObjectSubscript.hpp:64:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = unsigned char; TStringRef = const int&; typename ArduinoJson::Internals::JsonVariantAs<T>::type = unsigned char]'
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:63:35:   required from 'const typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = unsigned char; TImpl = ArduinoJson::JsonObjectSubscript<const int&>; typename ArduinoJson::Internals::JsonVariantAs<T>::type = unsigned char]'
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:58:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T() const [with T = unsigned char; TImpl = ArduinoJson::JsonObjectSubscript<const int&>]'
/Users/mrmld/Projects/MRobotics.Volt/MRobotics.Volt.CryptoContainer/MRobotics.Volt.CryptoContainer.ino:114:16:   required from here
/Users/mrmld/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:297:67: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'
       if (Internals::StringTraits<TStringRef>::equals(key, it->key)) break;
                                                                   ^
exit status 1
```

DiE-MaRMeLaDe, 2018-03-05:
It appears, when I try to pass JsonObject in arguments to function

bblanchon, 2018-03-05:
Hi @DiE-MaRMeLaDe,

Please have a look at:

* [FAQ: Error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'](https://arduinojson.org/faq/error-equals-is-not-a-member-of/?utm_source=github&utm_medium=issues)

Regards,
Benoit

DiE-MaRMeLaDe, 2018-03-07:
Thanks. It works

closed 2018-03-07 by DiE-MaRMeLaDe

bblanchon, 2018-03-08:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)