bblanchon/ArduinoJson #691: ArduinoJson  problem with Multiple object JSON response  - NodeMCU v1

HanyTalaatRamadan, 2018-03-06:
Hi ,
Thanks a lot for the great Library , it was very helpful for me, and i'm new IoT
i'm using NodeMCU v1 and I'm using ArduinoJson to parse JSON response from REST web service , the Web Service response contains multiple Objects in JSON as following

```
{
"items" : [
{
"ThingId" : 2,
"ThingName" : "Tank-CAT3406-1",
"PropertyName" : "Radius",
"PropertyValue" : "3.5",
"DeviceId" : 5,
"SensorId" : 2,
"ThingTypeId" : 2,
"Recid" : 1,
"links" : [
{
"rel" : "self",
"name" : "ThingPropertiesRO1",
"kind" : "item"
},
{
"rel" : "canonical",
"name" : "ThingPropertiesRO1",
"kind" : "item"
}
]
},
{
"ThingId" : 2,
"ThingName" : "Tank-CAT3406-1",
"PropertyName" : "Width",
"PropertyValue" : "10",
"DeviceId" : 5,
"SensorId" : 2,
"ThingTypeId" : 2,
"Recid" : 2,
"links" : [
{
"rel" : "self",
"name" : "ThingPropertiesRO1",
"kind" : "item"
},
{
"rel" : "canonical",
"name" : "ThingPropertiesRO1",
"kind" : "item"
}
]
},
{
"ThingId" : 2,
"ThingName" : "Tank-CAT3406-1",
"PropertyName" : "Hight",
"PropertyValue" : "5",
"DeviceId" : 5,
"SensorId" : 2,
"ThingTypeId" : 2,
"Recid" : 3,
"links" : [
{
"rel" : "self",
"name" : "ThingPropertiesRO1",
"kind" : "item"
},
{
"rel" : "canonical",
"name" : "ThingPropertiesRO1",
"kind" : "item"
}
]
}
],
"count" : 3,
"hasMore" : false,
"limit" : 5,
"offset" : 0,
"links" : [
{
"rel" : "self",
"href" : "http://127.0.0.1:7101/PowerSenseApp-RESTWebService-context-root/rest/beta/ThingPropertiesRO1",
"name" : "ThingPropertiesRO1",
"kind" : "collection"
}
]
}

```
when i'm trying to parse the response to multiple Json Objects it fails and return null , but for one object in response it works fine

```
void getTankInit(String stringResponse){
  const size_t bufferSize = 16580;//JSON_ARRAY_SIZE(1) + 3*JSON_ARRAY_SIZE(2) + JSON_ARRAY_SIZE(3) + 7*JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(6) + 3*JSON_OBJECT_SIZE(9) + 1748000;
  Serial.println(bufferSize);
  Serial.println(stringResponse.length());
DynamicJsonBuffer jsonBuffer(bufferSize);
JsonObject& root = jsonBuffer.parseObject(stringResponse);
Serial.println(root.size());
JsonArray& items = root["items"];
for (int i=0; i<items.size());i++) {
JsonObject& items0 = items[i];
const char* items0_ThingName = items0["ThingName"]; 
const char* items0_PropertyName = items0["PropertyName"]; 
const char* items0_PropertyValue = items0["PropertyValue"]; 

Serial.println(items0_PropertyName); Serial.println(items0_PropertyValue);
}
```
i appreciate your help for my case .

bblanchon, 2018-03-06:
Hi @HanyTalaatRamadan,

This program consumes a lot of memory and can probably be optimized.

First, the `String` parameter creates a complete copy of the input; so between 1KB and 2KB are lost, from what I can see.
Then, the `JsonBuffer` is way too big, the [ArduinoJson Assistant](https://arduinojson.org/assistant/?utm_source=github&utm_medium=issues) says that 2KB is enough.
To reduce the memory consumption further, you need to avoid the `String`.

See:

* [FAQ: Why parsing fails?](https://arduinojson.org/faq/why-parsing-fails/?utm_source=github&utm_medium=issues)
* [FAQ: How to reduce memory usage?](http://arduinojson.org/faq/how-to-reduce-memory-usage/?utm_source=github&utm_medium=issues)

Regards,
Benoit

HanyTalaatRamadan, 2018-03-07:
Thanks @bblanchon  for help ,
Yes i agree that String is consuming memory and can be replaced with char .
also i agreed that the size of `JsonBuffer` is big , it was set to contain the whole response and i will reduce it .

The Problem is 
in case of response contains multiple objects :
 JsonObject& root = jsonBuffer.parseObject(stringResponse); 
Serial.println(root.size()); returned size is 0 

and in case response contains single object : 
returned size is correct 

Thanks

bblanchon, 2018-03-08:
> in case of response contains multiple objects

What do you mean by that?
Is the input still a valid JSON document?
Did you wrap the objects in an array?
Is there enough memory left after the input is loaded?

> and in case response contains single object : returned size is correct

Could you explain?
What is the expected result?
What do you get instead?

See also:

* [FAQ: Can I parse a JSON input that is too big to fit in memory?](https://arduinojson.org/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/?utm_source=github&utm_medium=issues)

Regards,
Benoit

closed 2018-03-16 by bblanchon