bblanchon/ArduinoJson #693: Is is possible to parse a simple value into JsonVariant?

Adam5Wu, 2018-03-08:
Thanks for the great library! It saves me so much time and headache. :)

However, I am encountering some difficulties using the parse method for a slightly different purpose.
I am trying to write a utility which accepts user input and update json accordingly.

For example:
* Original json: `{ "Test": "abc" }`
* User input: Test="def"
* My logic intends to do the following:
  1. Parse input string and obtain the key `Test` and value `"def"`
  2. Use ArduinoJson to parse the value `"def"` into a JsonVariant, make sure user indeed provide a simple string, not a json object to array
  3. Assign the parsed JsonVariant to the original json with the given key
* Expected new json: `{ "Test": "def" }`

However, I am stuck on step 2, because when I call `jsonBuffer.parse("\"def\"", 0)` the result always fails.
Reading the document, I suspect the parse method can only accept serialized json object or array, not a simple value, such as boolean, integer, or string. Am i missing anything?

Note that since the utiltiy is general purpose, I cannot know ahead of time which field is which type, I think it is a perfect case to utilize the parser and JsonVariant to achieve this goal and keep the code simple and elegant.

bblanchon, 2018-03-08:
Hi @Adam5Wu,

You are right, this is a bug.

As a simple example:

```c++
jsonBuffer.parseArray("[]", 0); // works
jsonBuffer.parse("[]", 0); // fails
```

Thank you very much for reporting this issue.
I'll fix it as soon as possible.

Regards,
Benoit

Adam5Wu, 2018-03-08:
Hi @bblanchon, thanks for the prompt updates. However, I still found some percularity around nesting.

The documented behavior:
- With nesting limit of 0, a flat object or array can be decoded, e.g. `{"test":1}` and `[1,2,3]`
- With nesting limit of 1, an object or array can embed another flat object or array, e.g. `[{"test":1}]`

Actual code behavior is slight different:
- With nesting limit of 0, a simple value, or an empty object or array can be decoded, e.g. `"test"`, `{}` and `[]`
  - A non-empty object or array cannot be decoded
- With nesting limit of 1, a flat object or array can be decoded, and an empty object or array can be embeded, e.g `{"test":1}`, `{"test":[]}` and `[1,{},3]`

And I think none of the above behavior is coherent.
The code behavior is slightly closer to coherency than documented behavior.

Imagine the following scenario:
1. If simple values such as string and number, as well as flat object and array, all considered to have nesting level of 0;
2. Then the problem is, when a JsonVariant of nesting level 0 is nested inside another container (object or array), the result nesting level would be different!
    - Embedding simple value into a flat object/array will not change the nesting level;
        - Nesting level algebra: 0+0 = 0
    - Embedding flat object/array into a flat object/array will increase the nesting level
        - Nesting level algebra: 0+0 = 1

I think to solve this conflict, the concept of nesting needs to be tweaked a little bit.
- Nesting level 0 would only map to simple values, never object or array
- Each layer of object/array will increase nesting level by 1

What do you think?

Adam5Wu, 2018-03-09:
I have attempted a fix in PR #695

bblanchon, 2018-03-09:
Good job!
Thanks.

bblanchon, 2018-03-09:
Humm, there is something bothering me...

How do you justify this:

```c++
SHOULD_WORK(jb.parseArray("[]", 0));
SHOULD_FAIL(jb.parse("[]", 0));
```

It doesn't make sense.

Too me, they should both fail

Adam5Wu, 2018-03-09:
To me, I feel `parse` and `parseArray/parseObject` are two classes of APIs.
- `parse` is a general purpose API, it can parse anything, simple value, or composites
  - Therefore, nest limit should be enforced from level 0
- `parseArray/parseObject` are special purpose APIs, their are made to parse a composite value
  - Therefore, implicitly, people who use these API already expect nest level at least 1

Then, it is up the the interpretation of the nest_limit parameter:
- If it is defined as a "delta" value (how many additional level of nesting from the "expected base"), then the following makes sense:
  ```
  SHOULD_WORK(jb.parseArray("[]", 0));
  SHOULD_FAIL(jb.parse("[]", 0));
  ```
- If it is defined as an absolute nest level, then your latest amendments makes sense.

I personally am OK for either interpretations.

But I guess for backward compatibility, the delta interpretation may cause less complaint. (Say, some projects have used parseObject or parseArray according to old documentation, they may be surprised that `jb.parseArray("[1,2,3]", 0)` stopped working)

commit efea6b2f (2018-03-09): Fixed inconsistencies in nesting level counting (fixes #693)
commit 3fcd9405 (2018-03-12): Fixed `JsonBuffer::parse()` nesting limit correctly (fixes #693)
commit 3523296e (2018-03-12): Fixed `JsonBuffer::parse()` nesting limit (fixes #693)
closed 2018-03-12 by bblanchon