bblanchon/ArduinoJson #694: Can't parse json from server

AL3X1, 2018-03-08:
I send request from server to arduino and receive response. In server (with php) i can't decode json, i don't know why. Online services said what json is valid. Where is my problem?

**Code from Arduino:** 

    BufferFiller bfill;
    StaticJsonBuffer<200> jBuffer;
    JsonObject& root = jBuffer.createObject();

	char ipAddr[20];
	sprintf(ipAddr, "%d.%d.%d.%d", ether.myip[0], ether.myip[1], ether.myip[2], ether.myip[3]);

	root["name"] = "Arduino";
	root["ip"] = ipAddr;

	String response = "";
	root.printTo(response);

	bfill = ether.tcpOffset();
	bfill.emit_p(PSTR(
		"HTTP/1.0 200 OK\r\n"
		"Content-Type: text/html\r\n"
		"\r\n"
		"<title>Identity response</title>"
		"$S"), response.c_str());`

Arduino's response: `string(80) "{"name":"Arduino Mega 2560","ip":"192.168.1.20"}"`
**Interesting moment:**
 When i generate this json from php `var_dump(json_encode(["name" => "Arduino Mega 2560", "ip" => "192.168.1.20"]));`
I receive output: `string(**48**) "{"name":"Arduino Mega 2560","ip":"192.168.1.20"}"`

I think problem here, in size of json, but how i can fix it??

bblanchon, 2018-03-09:
Hi @AL3X1,

Maybe it's just the `Content-Type` that needs to be set to `application/json`.

Regards,
Benoit

AL3X1, 2018-03-14:
Problem was in `<title>Identity response</title>`. Here is 32 symbols :D I just delete this and decoding is ok. Thanks ;-)

closed 2018-03-14 by AL3X1

bblanchon, 2018-03-15:
You're welcome @AL3X1. 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)