bblanchon/ArduinoJson #696: Floating Point Precision edge case

terryjmyers, 2018-03-09:
Of course I googled this and there is a plethora of answers regarding floating point precision and scientific notation, but I think I've found an edge case.

I'm using a 32-bit ESP32 as the micro controller and saving calibration data as an array of floats to a json on SPIFFS.  Everything works great except when you have a number that is almost scientific notation.  For example the number 3.57848E-05 is encoded as 0.000036 in the json losing valuable precision.
Along the same lines I'm unable to save more than 8 digits (including decimal place).  I really don't need more than 8 digits of precision for my application, but the library seems to truncate to 8 digits even if its encoding a double data type (which would be 64bits of data in my microcontroller) or if explicitly  specifying the number of digits in the .add member(i.e jsonADC1CalibrationData.add(ADC1CalibrationData[0], 15);)

bblanchon, 2018-03-11:
Hi @terryjmyers,

Make sure you set [ARDUINOJSON_USE_DOUBLE](https://arduinojson.org/api/config/use_double/) to 1 to enable 64-bit precision.
You can also adjust `ARDUINOJSON_NEGATIVE_EXPONENTIATION_THRESHOLD` to trigger exponentiation earlier.

See demo here: https://wandbox.org/permlink/UrYasUGFHEnL9aVU

Regards,
Benoit

terryjmyers, 2018-03-12:
@bblanchon 
Yes I had defined the ARDUINOJSON_USE_DOUBLE after I posted that and this seemed to solve my problem temporarily, but I was still worried about if I had a different number closer to the exponent threshold for even a double.  But I DID NOT know about theARDUINOJSON_POSITIVE_EXPONENTIATION_THRESHOLD and ARDUINOJSON_NEGATIVE_EXPONENTIATION_THRESHOLD.  This is what I really needed

So just to ensure I always save the maximum amount of precision, I'm just going to set both to 1:
#define ARDUINOJSON_POSITIVE_EXPONENTIATION_THRESHOLD 1
#define ARDUINOJSON_NEGATIVE_EXPONENTIATION_THRESHOLD 1

Thanks again!

closed 2018-03-12 by terryjmyers

bblanchon, 2018-03-12:
Indeed, there were missing from the documentation.
Here there are now:

* [ARDUINOJSON_POSITIVE_EXPONENTIATION_THRESHOLD](https://arduinojson.org/api/config/positive_exponentiation_threshold/?utm_source=github&utm_medium=issues)
* [ARDUINOJSON_NEGATIVE_EXPONENTIATION_THRESHOLD](https://arduinojson.org/api/config/negative_exponentiation_threshold/?utm_source=github&utm_medium=issues)