bblanchon/ArduinoJson #697: JsonBuffer persistance: adding to JSON object in intervals

Armannas, 2018-03-09:
Arduino IDE: 1.8.6
Device: ESP8266

Hi there,

I am doing ADC measurements in batches/intervals and would like to send stuff over MQTT only after n batches. In the naive case I would build a huge array(adding all measurements of all intervals to it) before adding all values to an ArduinoJSON object. To make it more memory efficient, I would like to use a buffer array that is just big enough for one interval and then intermittently add to the JSON object. In the next iteration, I would then overwrite the buffer and repeat for a few intervals. 

If I understand correctly, the JsonBuffer is not persisent over function calls, so I need to do everything I want in the function where I construct the JsonBuffer. Would the best way to do this, to create a dummy function where my code lives for the given amount of intervals like so?:

```c
void loop () {
    // JsonBuffer lives in main(). Destructed after 4 batches.
    void main(void) {
        StaticJsonBuffer<SOME_BUFFER_SIZE> jsonBuffer; 
       JsonObject& root = jsonBuffer.createObject();
       JsonArray& jsonVoltage = root.createNestedArray("v"); // v = voltage measurements
        

        // Repeat for 4 batches/intervals
        while( j < 4) {
            // Other things my program might want to do here

            for(int i =0; i<100; i++) {
                // ADC measurements here
                measurement[i] = . . .
            }
            // JSON array population for this interval
            for(int i = 0;i<100;i++) {
                jsonVoltage.add(measurement[i]);
            }
        j++
       }
      
       // Reset batch measurement counter
       j = 0

       char buff[SOME_BUFFER_SIZE];
       root.printTo(buff,sizeof(buff)); 
   
       // Send through MQTT and do other stuff the program might want to do
    }
    // Exited main function. JsonBuffer is destructed and a new one is created 
    // when re-entering the loop
}

```

It might be a trivial question, but I would just like to know what the cleanest option would be.

Thanks.

PS:
Also thanks for the nice library, very motivational to dive into C++ more.

bblanchon, 2018-03-12:
Hi @Armannas,

>  To make it more memory efficient, I would like to use a buffer array that is just big enough for one interval and then intermittently add to the JSON object.

This is an excellent idea.

> Would the best way to do this, to create a dummy function where my buffer (and some code) lives for the given amount of intervals like the code below? Or can I just create the jsonbuffer in the loop(), such that the jsonbuffer is destructed and constructed after exiting and re-entering the loop(), respectively?

Both would work, but with a function, you get the opportunity to give a name to the portion of the code, which makes it more readable.
So, to me, the cleanest way is use a function.
As Uncle Bob says "Extract till you drop" :-)

Regards,
Benoit

Armannas, 2018-03-14:
Thanks for the clear answers and feedback, monsieur!

closed 2018-03-14 by Armannas

bblanchon, 2018-03-15:
You're welcome @Armannas. 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)