bblanchon/ArduinoJson #698: Passing JsonObject& reference

rayozzie, 2018-03-11:
I apologize in advance because this is almost certainly a C++ question, and although I hesitated to post here, I'm thinking that someone else will ultimately be searching for the same answer.  My thanks for your patience.

I am using ArduinoJson to process json requests/responses to a device over serial.  I have one source file that encapsulates >25 different requests, with each C-friendly subroutine of the form:

    bool deviceGetStatus(char *statusBuf, int statusBufLen) {

        // Prepare the request
        StaticJsonBuffer<deviceTransactionDefaultSize> json;
        JsonObject& req = json.createObject();
        req["req"] = "device.status";

        // Perform the transaction and check for valid response
        xSemaphoreTake(reqMutex, portMAX_DELAY);
        req.printTo(Serial2);
        Serial2.println();
        Serial2.flush();
        JsonObject& rsp = json.parse(Serial2);
        xSemaphoreGive(reqMutex);
        if (!rsp.success())
            return false;

        // Check 
        if (!rsp["status"].success)
            return false;
        const char *status = rsp["status"];
        strncpy(statusBuf, status, statusBufLen-1);
                                                                                            
        return true;

    }

As you can see, ArduinoJson is wonderful in that it enables an extremely natural syntax, and I appreciate all you've done to enable this.

However, because I've got 25+ functions such as this, I'm trying to simplify it even more, for maintenance purposes.  As such, I'd like to make the "middle part", which is "Perform the transaction and check for valid response", a subroutine.  So, with zero knowledge of C++ and a robust knowledge of C, I tried the following.  (I also tried many other things, but I was stumbling around in the dark.  I'm laying it out this way for clarity.)

There are two classes of error that arise from this, one that is clearly my lack of understanding of C++ and the other being something related to how classes are laid out in ArduinoJson (because "parse" could not be found).  

Might I trouble you to assist me, and others in the future who attempt such a thing, by giving the syntax for the transaction function that would enable me to achieve what I'm setting out to do?

Thank you much in advance, and thanks again for ArduinoJson.

req.cpp:82: error: cannot declare pointer to 'class ArduinoJson::JsonObject&'
 bool transaction(JsonBuffer &json, JsonObject& req, JsonObject& *prsp) {
                    (^ is under "*prsp")

req.cpp: In function 'bool transaction(ArduinoJson::JsonBuffer&, ArduinoJson::JsonObject&, ArduinoJson::JsonObject*)':
req.cpp:85: error: 'class ArduinoJson::JsonBuffer' has no member named 'parse'
     JsonObject& rsp = json.parse(Serial2);
                   (^ is under "parse")

req.cpp:88: error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
     *prsp = rsp;
                   (^ is under "=")

    bool transaction(JsonBuffer& json, JsonObject& req, JsonObject& *prsp) {
        xSemaphoreTake(reqMutex, portMAX_DELAY);
        req.printTo(Serial2);
        Serial2.println();
        Serial2.flush();
        JsonObject& rsp = json.parse(Serial2);
        xSemaphoreGive(reqMutex);
        if (!rsp.success())
            return false;
        *prsp = rsp;
        return true;
    }

    bool deviceGetStatus(char *statusBuf, int statusBufLen) {

        // Prepare the request
        StaticJsonBuffer<deviceTransactionDefaultSize> json;
        JsonObject& req = json.createObject();
        req["req"] = "device.status";

        // Perform the transaction and check for valid response
        if (!transaction(json, req, &rsp))
            return false;

        // Check 
        if (!rsp["status"].success)
            return false;
        const char *status = rsp["status"];
        strncpy(statusBuf, status, statusBufLen-1);
                                                                                            
        return true;

    }

bblanchon, 2018-03-12:
Hi  @rayozzie,

Here is a possible solution:

```c++
JsonObject transaction(JsonBuffer& json, const JsonObject& req) {
    xSemaphoreTake(reqMutex, portMAX_DELAY);
    req.printTo(Serial2);
    Serial2.println();
    Serial2.flush();
    JsonObject& rsp = json.parse(Serial2);
    xSemaphoreGive(reqMutex);
    return rsp;
}

bool deviceGetStatus(char *statusBuf, int statusBufLen) {
    // Prepare the request
    StaticJsonBuffer<deviceTransactionDefaultSize> json;
    JsonObject& req = json.createObject();
    req["req"] = "device.status";

    // Perform the transaction
    JsonObject& rsp = transaction(json, req);

    // Check 
    const char *status = rsp["status"];
    if (!status) // the pointer will be null is the object is invalid or if "status" is missing
       return false;
    strlcpy(statusBuf, status, statusBufLen-1);
                                                                                        
    return true;
}
```

rayozzie, 2018-03-12:
Thanks for your constructive reply.  Appreciated.

That leaves this problem pending; it can't seem to find the parse method.

sketch/box.cpp: In function 'ArduinoJson::JsonObject& boxTransaction2(ArduinoJson::JsonBuffer&, ArduinoJson::JsonObject&)':
box.cpp:86: error: 'class ArduinoJson::JsonBuffer' has no member named 'parse'
     JsonObject& rsp = json.parse(SerialBox);
                            ^

rayozzie, 2018-03-12:
I should also mentioned that I tried this and it failed.  I'm on 5.13.1.

sketch/box.cpp: In function 'ArduinoJson::JsonObject& boxTransaction2(ArduinoJson::JsonBuffer&, ArduinoJson::JsonObject&)':
box.cpp:86: error: 'class ArduinoJson::JsonBuffer' has no member named 'parseObject'
     JsonObject& rsp = json.parseObject(SerialBox);
                            ^

rayozzie, 2018-03-12:
Aaah ... good news.  I found the answer in here:
https://github.com/bblanchon/ArduinoJson/issues/414
https://github.com/bblanchon/ArduinoJson/releases/tag/v5.8.0

As it turns out, of the different proposed variants of function definition, this is the only one that worked:
template <typename TJsonBuffer> JsonObject& boxTransaction(TJsonBuffer &json, JsonObject& req) {

That said, it does indeed work.

Thanks again.

closed 2018-03-12 by rayozzie

bblanchon, 2018-03-13:
You're welcome.
Thank you for using ArduinoJson.