bblanchon/ArduinoJson #701: Send Json Object to Server Get Request

rizban114406, 2018-03-12:


rizban114406, 2018-03-12:
#include <ArduinoJson.h>
int8_t answer;
void setup() {
  Serial.begin(9600);
  setNetworkGSM();

}

void loop() {
  char body[200];
  char response[100];
  DynamicJsonBuffer jBuffer;
  JsonObject& root  = jBuffer.createObject();
  JsonObject& root2  = jBuffer.createObject();
  JsonArray& data  = root.createNestedArray("data");
  root2["userid"] = 1;
  root2["trip_reference"] = 12434243;
  root2["latitude"] = "23.7907832";
  root2["longitude"] = "90.4090195";
  root2["date"] = "20160510";
  root2["time"] = "12:10:02";
  root2["speed"] = "100";
  root2["distance"] = "100";
  data.add(root2);
  root.printTo(body);
  //Serial.println(body);
  
  doHttpRequests(body,response);
  Serial.println(response);
}
void setNetworkGSM()
{
  delay(10000);
  while (sendATcommand("AT+CREG?", "+CREG: 0,1", 2000) == 0);
  sendATcommand("AT+SAPBR=3,1,\"Contype\",\"GPRS\"", "OK", 2000);
  sendATcommand("AT+SAPBR=3,1,\"APN\",\"internet\"", "OK", 2000);
  while (sendATcommand("AT+SAPBR=1,1", "OK", 20000) == 0)
  {
    delay(5000);
  }
}
void doHttpRequests(char* body, char* response)
{ 
  answer = sendATcommand("AT+HTTPINIT", "OK", 10000);
  //String data = body;
  if (answer == 1)
  {
    answer = sendATcommand("AT+HTTPPARA=\"CID\",1", "OK", 5000);
    if (answer == 1)
    {
      char aux_str[300];
      sprintf(aux_str, "AT+HTTPPARA=\"URL\",\"http://Server_IP_Address/carcopolo_arduino/Json_index.php?tag=%s&data=%s","insertdevicedata",body);
      Serial.print(aux_str);
      answer = sendATcommand("\"", "OK", 10000);
      if (answer == 1)
      {
        answer = sendATcommand("AT+HTTPACTION=0", "+HTTPACTION: 0,200", 15000);
        if (answer == 1)
        {
          answer = sendATcommand("AT+HTTPREAD", "+HTTPREAD:", 35000);
          
          if(answer == 1)
          {
            readResponse(response,10000);
          }
        }
      }
    }
  }
  sendATcommand("AT+HTTPTERM", "OK", 5000);
}

Here is my code. I have made a json object and copied it to a character array named body to send it to the server via get request. I have checked the serial monitor and it prints perfectly. 
AT+HTTPPARA="URL","
http://Server_IP_Address/carcopolo_arduino/Json_index.php?tag=insertdevicedata&data={
"data":[{"userid":1,"trip_reference":12434243,"latitude":"23.7907832","longitude":"90.4090195","date":"20160510","time":"12:10:02","speed":"100","distance":"100"}]}"

But I am getting an Error response. I have copied the url and tested it in a browser and it worked fine. Can you please tell me what mistake i have done that i am getting this error??? thanks in advance.

bblanchon, 2018-03-13:
Hi @rizban114406,

I'm sorry, there are too much code and too little information.
Please provide an [MCVE](https://stackoverflow.com/help/mcve), rephrase your question, and we'll find the problem.

See also:

* [FAQ: Why parsing fails?](https://arduinojson.org/faq/why-parsing-fails/?utm_source=github&utm_medium=issues)

Regards,
Benoit

closed 2018-03-21 by bblanchon