bblanchon/ArduinoJson #702: Problems with ArduinoJson and Ethernet Libraries

AvatarRolon, 2018-03-12:
Excuse me, my doubt is that ArduinoJson is actually compatible with Ethernet Library, because i can not send data from arduino ethertnet to my server.
Explanation:
The object json is pretty well formed only when the variables for ethernet are not present, but when variables for ethernet are present, the object json is not formed as well as without variables, can you tell me why is passing this?
example with out variable for ethernet:
{
  "coordinador": "013A20040D95BE0",
  "dispositivo": "013A20040B7D712",
  "fecha": "12/12/12",
  "hora": "12:12:12,
  "sensores": [
    {
      "nombre": "DHT",
      "datos": {
        "temperatura": 10.5,
        "humedad": 10.5
      }
    },
    {
      "nombre": "Higometro1",
      "datos": {
        "humedad": 273
      }
    },
    {
      "nombre": "Higometro2",
      "datos": {
        "humedad": 254
      }
    },
    {
      "nombre": "Higometro3",
      "datos": {
        "humedad": 263
      }
    },
    {
      "nombre": "Higometro4",
      "datos": {
        "humedad": 281
      }
    }
  ]
}

With ethernet variables
{
  "coordinador": "013A20040D95BE0",
  "dispositivo": "013A20040B7D712",
  "fecha": "165/165/2165",
  "hora": "165:165:85",
  "sensores": [
    {
      "nombre": "DHT",
      "datos": {
        "temperatura": NaN,
        "humedad": NaN
      }
    },
    {
      "nombre": "Higometro1",
      "datos": {
        "humedad": 273
      }
    },
    {
      "nombre": "Higometro2",
      "datos": {
        "humedad": { }
    }
  ]
}

Sure i can give you my code, if it help, thanks

bblanchon, 2018-03-13:
Hi @AvatarRolon,

I've been using ArduinoJson with the Ethernet library from the beginning, so I can testify that it works.

I guess that you don't have enough memory on your device.

See:

* [FAQ: Why are some parts missing?](https://arduinojson.org/faq/why-some-parts-are-missing/?utm_source=github&utm_medium=issues]

Regards,
Benoit

AvatarRolon, 2018-03-13:
Thanks, i did what the FAQ POST says and it works, i'm too grateful for your assistance.

bblanchon, 2018-03-13:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this project [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)

closed 2018-03-13 by bblanchon