bblanchon/ArduinoJson #704: Non zero-terminated char buffer / Zero-copy / Input buffer length

ilyary, 2018-03-14:
I'm parsing data from ssl_read. The data is char sequence but not zero-terminated, though length is known. I use zero-copying. There is a threat to go out of buffer range when braces are not closed or JSON corrupted.
How to make use of known input buffer length?

bblanchon, 2018-03-14:
Hi @ilyary,

Unfortunately, there is no direct method to do that.

Your options are:

* Add a zero at the end of your input
* Wrap the input in a stream ([see an example here](https://github.com/bblanchon/ArduinoJson/blob/master/fuzzing/fuzzer.cpp#L20))

Regards,
Benoit

ilyary, 2018-03-14:
"Wrap the input in a stream" seems not be zero-copy. True?

bblanchon, 2018-03-14:
There is no copy, but you pay the cost of a virtual call.
If I were you, I would just append the zero.

Anyway, I think in the future we must support `string_view`.

ilyary, 2018-03-14:
Then input in a stream is nice.

I use ESPAsyncTCP and buffer is allocated in system underlying function ssl_read, then ESPAsyncTCP frees it. So, I see no possibility to extend the buffer by 1.

But I'm new to c/cpp language and libraries, so maybe there is some way to do it without copying or maybe the buffer is already allocated +some bytes))

Assuming that I gonna use stream.

Thank you for prompt reply

Best regard,
Ilya

bblanchon, 2018-03-15:
Humm. You're right; I got confused.
The stream will make ArduinoJson switch to the "normal" mode, where it copies the relevant bits of the input.
But, it may be what you want since, as you said, ESPAsyncTCP frees the buffer.

Too bad you cannot add one byte to the array, because, from a performance perspective, it would be the best.

Anyways, I said, we need to support string_view in ArduinoJson.
The only problem is that I'm currently focused on v6 so I do not want to add new features to v5.

ilyary, 2018-03-15:
Good luck! ArduinoJson is really helpful!

bblanchon, 2018-06-07:
This feature is available in [ArduinoJson 6.0.0](https://arduinojson.org/news/2018/06/07/version-6-0-0/?utm_source=github&utm_medium=issues)

commit ccb54136 (2018-05-14): Added support for non zero-terminated strings (fixes #704)
closed 2018-06-07 by bblanchon