bblanchon/ArduinoJson #705: Unexpected gap in required StaticJsonBuffer

Armannas, 2018-03-14:
Wemos D1 R2 & Mini (ESP8266)
Arduino 1.8.6

I am trying to process large arrays in JSON. An example below:

```json
{"d":{"ID":"Farm1","Ts":15,"v":[0.134,...,0.134],"t":[1520717223,....,1520717223]}}
```
Where the dots represent repetition. The array size of  key 'v'(voltage) is a multiple of 100 and every 100 voltages have one timestamp in key "t". For 100 measurements the document looks like this:
```json
{"d":{"ID":"test","Ts":15,"v":[0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134,0.134],"t":[1520717223]}}
```
According to the Assistant I needed a size of 1956 for my buffer. Experimentally I found 1330 worked as well.

Trying to scale this to 200 measurements (200 voltages, 2 timestamps), the assistant suggested a size of 3779, however experimentally I got stack memory errors, unless I chose a buffer size of at least ~16000!

For 300 measurements, the assistant suggested 5602 and experimentally something like ~18000 worked.

What is happening that is causing this huge jump followed by normal increments?

Generated the situation with the code below:

```c
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson/releases/tag/v5.0.7

#define DEVICE_ID "test1"

int publishSize = 1;
int batchSize = 100;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); delay(2000); debugPrintln();
}

void loop() {
  // put your main code here, to run repeatedly:

  publishData();
  delay(100);
  Serial.println("iteration done");
  delay(10000);
}

void publishData() {
 // Using https://arduinojson.org/assistant/ to give initial buffersize estimation
 StaticJsonBuffer<1330> jsonBuffer; // ~1330 for batch of 100
 Serial.println("JSON buffer created");
 JsonObject& root = jsonBuffer.createObject();
 JsonObject& d = root.createNestedObject("d");
 // Device ID
 d["ID"] = DEVICE_ID;
 
 // Sampling interval
 d["Ts"] = 15;
 
 JsonArray& jsonVoltage = d.createNestedArray("v"); // v = voltage
 JsonArray& jsonTimestamp = d.createNestedArray("t"); // t = unix timestamp
 
 for (int j = 0; j < publishSize;j++) {
   jsonTimestamp.add(1520717223);
  for(int i = 0;i<batchSize;i++) {
    jsonVoltage.add(0.134);
  }
 }
 
 Serial.println("JSON structure created");
 char buff[root.measureLength()+1];
 
 Serial.println(sizeof(buff));
 Serial.println("Character buffer created");

 Serial.println("writing to buffer");
 root.printTo(buff,sizeof(buff));
 Serial.println(buff);
 Serial.println("wrote to buffer");
 }

bblanchon, 2018-03-15:
Hi @Armannas,

```c++
StaticJsonBuffer<16000> jsonBuffer;
```

Unless your running on a PC, this is way too big for the stack.
What you're experiencing is probably due to a corruption of the stack.
Switch to a `DynamicJsonBuffer` and everything should be fine.

About the Assitant: in the table, it includes "Additional bytes for input duplication".
These bytes are only needed when ArduinoJson duplicates the string: either because it's parsing a read-only input, or because you construct a `JsonObject` with volatile strings.
In your case, you don't need to add the "Additional bytes for input duplication".

Also, please tell me you're not actually using v5.0.7 ;-)

Regards,
Benoit

closed 2018-03-22 by bblanchon

Armannas, 2018-04-16:
Your comments exactly solved my problem. I had to educate a bit on when stuff is put on the stack and the heap.

Thanks.

bblanchon, 2018-04-17:
You're welcome.
Thanks for the update @Armannas.