bblanchon/ArduinoJson #706: Utility Function to reduce code

chadham, 2018-03-15:
Hello. Awesome library, thank you for sharing.

Quick question. I find I've got 30+ references to the following code:

_DynamicJsonBuffer jsonBuffer;
JsonObject& reply     = jsonBuffer.createObject();
JsonObject& outP      = reply.createNestedObject(name);_

with the only difference being the name of the nestedObject.

Tried to create a method that took the name of the nestedObect as a parameter and returned a stf::pair of JsonObjects but couldn't get it to compile (quite possibly to my newness to C++).

Is there a simple way to create a utility function to replace these 3 lines with one line of code? I need handles to the reply and out objects downstream.

Thank you

bblanchon, 2018-03-16:
Hello @chadham,

It's not that simple.
You cannot just replace the three lines with a function because we need a way to handle the lifetime of the `jsonBuffer` (remember: it hold the memory of every `JsonObject`s).

Your only option is to wrap in an object.
Something like this:

```c++
class ReplyObj {
public:
  ReplyObj(const char* name) : _obj(_jb.createObject()), _outP(_obj.createNestedObject(name))
  {}

  JsonObject& outP() { return _outP; }

private:
  DynamicJsonBuffer jsonBuffer;
  JsonObject& _obj;
  JsonObject& _outP;
};
```

But I really think it complicates things.

If I had such a repetition in one of my programs, I would try to factor at a higher level of abstraction, not at the JSON serialization layer.

Regards,
Benoit

PS: you can see an more detailed example in the "case studies" of [Mastering ArduinoJson](https://arduinojson.org/book/?utm_source=github&utm_medium=issues);

chadham, 2018-03-16:
Thank you for your insights.

I currently declare the dynamic buffer each time, and assume the memory it uses is returned when it goes out of scope. Is my understanding correct? (Am new to C++).

Is there a benefit of using a shared buffer object?

closed 2018-03-26 by bblanchon