bblanchon/ArduinoJson #707: Issues with reading JSON file when using 4G tethered or dongle

mars000, 2018-03-16:
Hi

I am using the exact same code below. When I connect to via my local WiFi access point parsing is successful.
When I cannot via my iPhone or Android in 4G tethered mode parsing fails.

Here is the code and the output. I dont' know why this would be occuring. Any help would be appreciated.
```

bool readResponseContent(struct json_file_data *jsondata, WiFiClient *client_json_local)

{
	// Compute optimal size of the JSON buffer according to what we need to parse.
	// This is only required if you use StaticJsonBuffer.

	DynamicJsonBuffer jsonBuffer;

	Serial.println(F("waiting for parsing to occur..."));

	JsonObject &root = jsonBuffer.parseObject(*client_json_local); // was *client_json

	if (!root.success())
	{

		Serial.println(F("** JSON parsing failed!"));

		return false;
	}
	else
	{

		Serial.println(F("Parsing successful !"));

	}
```

Output below : my loops tries 5 times then gives up, note files and heads seem to be read in correctly.
```

...........WL_CONNECTED = 3
WiFi status = 3
WiFi connected
IP address: 
192.168.43.132
 
Connect to xxxx.xxx.com
Connected
GET /xxx/xxx/csr_master.json
headers read correctly :-)
waiting for parsing to occur...
** JSON parsing failed!
**** WARNING: Failed to read json file ! Error 0x05
Disconnect from client_json
Connect to xxx.xxx.com
Connected
GET /xxx/xxx/csr_master.json
headers read correctly :-)
waiting for parsing to occur...
** JSON parsing failed!
**** WARNING: Failed to read json file ! Error 0x05
Disconnect from client_json
Connect to xxx.xxx.com
Connected
GET /xxx/xxx/csr_master.json
headers read correctly :-)
waiting for parsing to occur...
** JSON parsing failed!
**** WARNING: Failed to read json file ! Error 0x05
Disconnect from client_json
Connect to xxx.xxx.com
Connected
GET /xxxx/xxx/csr_master.json
headers read correctly :-)
waiting for parsing to occur...
** JSON parsing failed!
**** WARNING: Failed to read json file ! Error 0x05
Disconnect from client_json
Connect to xxx.com
Connected
GET /xxx/xxx/csr_master.json
headers read correctly :-)
waiting for parsing to occur...
** JSON parsing failed!
**** WARNING: Failed to read json file ! Error 0x05
Disconnect from client_json
```

mars000, 2018-03-16:
I did a bit more reading and included increased my http_timeout from 5000 to 12000 and it seemed to fix the problem.
Now I still trying to work out why my Blynk server is giving me a "Login Timeout" error ONLY when i use my Vodafone 4G dongle - anyone know what could be casuing it ;-)
```

    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.5.0 on Arduino

[14996] Connecting to xxx.xxx.com:80
[15047] <[02|00|01|00] e01xxxxxxxxxxxxxxxxxa2
[17049] Login timeout
```

bblanchon, 2018-03-17:
Hi @mars000,

Indeed, the timeout is the first thing that came to my mind.
I'm glad you found the answer by yourself.

I'm sorry I cannot help your next problem.
Good luck.

Regards,
Benoit

closed 2018-03-17 by bblanchon